prusa3d/PrusaSlicer #4682: When reloading a model that had manifold errors, warning triangle is still shown

timur-tabi, 2020-08-29:
### Version
2.2.0+linux-x64

### Operating system type + version
Kubuntu Linux 20.04

### 3D printer brand / version + firmware version (if known)
Prusa MK3s

### Behavior
When a model has manifold errors, the Plater view shows the number of errors as well as an exclamation mark next to the STL filename.

I then correct the manifold errors in the STL and replace the filename.  I think right-click on the model and select "Reload from disk".  This will replace the current model with the updated STL file.

Since there are no more manifold errors, the bottom of the info panel says, "Manifold: Yes".  However, the exclamation mark in the triangle remains.  In order to get rid of the triangle, I need to delete and reload the model.

This is obviously a cosmetic error, but it should be an easy fix.
[image]
[image]

lukasmatena, 2024-02-19:
SPE-2168

FidelCapo, 2024-04-30:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2024-04-30 by FidelCapo