prusa3d/PrusaSlicer #4684: slicer crashed when object is placed on single edge

KubaMiszcz, 2020-08-30:
### Version
Version 2.3.0-alpha0

### Operating system type + version
Build: PrusaSlicer-2.3.0-alpha0+UNKNOWN
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 25,729MB

### Behavior
slicer crashed when object is placed on single edge (area=0?) caused by: 
- implies from object origin
- after rotating object without paying attention
- when upload object with such origin and background processing is enabled

* Expected Results_
message box with warning?

lukasmatena, 2020-08-30:
I could not reproduce it, but maybe it could be same as #4647. Could you try compile and run branch `dk_notifications` to see if the issue disappears? If it doesn't could you build debug mode and provide a backtrace? Thanks.

lukasmatena, 2020-12-19:
Is this still an issue with [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1)?

KubaMiszcz, 2020-12-21:
nope, after upgrade it works excellent now

closed 2020-12-21 by KubaMiszcz