prusa3d/PrusaSlicer #4687: Wrong temp during filament switch

mbordoni, 2020-08-31:
### Version
2.2.0 VC01

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3S MMU2S latest formware 3.9.0

### Behavior
All the filaments have the same profile. After correctly doing dribbling and unloading the old filament, once loaded the new one it is set 230° even if any config is set about it.
In addition to this the printer rises the temp very slowly without succeeding to match it. (while if I set the temp at the boot I do not have any problem like this, so it is not a hw problem)

Please take a look to the dribbilng procedure to guess where the problem is.

rtyr, 2020-08-31:
Are you sure you have opened this issue in the correct repository? You should probably contact the author of the dribbling feature about this.

rtyr, 2020-08-31:
I believe that answer in #3045 was actually meant for this issue so I am closing this.

closed 2020-08-31 by rtyr