prusa3d/PrusaSlicer #4688: Prusaslicer crashing with custom build plate or svg file

JohnyRein, 2020-08-31:
### Version

PrusaSlicer 2.2.0+win64

### Operating system type + version

Windows 10 Home build 19041.450

### Behavior
* Prusaslicer randomly crash with custom build plate or svg file. PrusaSlicer crash more often with STL loaded. Less often with Texture only. Plus the STL file of the bed is misaligned with texture
* 1. Load custom bed shape and texture to ''Bed Shape'' settings.
   2. Save
   3. Close PrusaSlicer
   4. Open Prusaslicer
   5. Prusaslicer will most possibly crash immediately after start.
* Not crash.
* Prusaslicer work fine without custom Texture and custom build plate
[image]
[image]

#### Project File (.3MF) where problem occurs

[ender 3 bed.zip](https://github.com/prusa3d/PrusaSlicer/files/5150260/ender.3.bed.zip)

Thank you in advance for help.

rtyr, 2020-08-31:
It seems, that your STL is incorrectly positioned. 

Fixed STL, which works fine:
[Bed_235_235_v2.zip](https://github.com/prusa3d/PrusaSlicer/files/5152180/Bed_235_235_v2.zip)

[image]

JohnyRein, 2020-08-31:
Thank you for fixing the STL file. Still crashing with svg file.
[image]
[image]

rtyr, 2020-08-31:
I am unable to reproduce it, it works fine on my PC.

JohnyRein, 2020-08-31:
Then try to open this project. When model will show on the print bed, immediately click on Add...
[5015_30deg_Single_Radial_Fan_Dual_Ducts.zip](https://github.com/prusa3d/PrusaSlicer/files/5152941/5015_30deg_Single_Radial_Fan_Dual_Ducts.zip)

rtyr, 2020-08-31:
It is probably specific to your machine.

JohnyRein, 2020-08-31:
No. A friend with different model of laptop has crashes too.

rtyr, 2020-08-31:
Please provide system info of both of those PCs.

JohnyRein, 2020-08-31:
Dell Inspiron 7577

i7 7700HQ
16GB RAM
GTX 1060 maxQ
Windows 10 Home 64 bit

Dell G5 5590

i7 8750H
16 GB RAM
RTX 2060
Windows 10 Home 64 bit

enricoturri1966, 2020-09-01:
@JohnyRein
Could you please post the texture files you are using in svg format (the ones already attached are all in png format).
Svg files can be saved in many different ways and not all of them are compatible with PrusaSlicer (try for example to open in a text editor the svgs saved into resources\profiles\PrusaResearch)

JohnyRein, 2020-09-01:
Here is SVG file (saved in Illustrator) which is not working properly
[Ender-Grid_svg.zip](https://github.com/prusa3d/PrusaSlicer/files/5156919/Ender-Grid_svg.zip)
[image]

and PSD file with layers used for create the bed texture.
[Ender-Grid.zip](https://github.com/prusa3d/PrusaSlicer/files/5156923/Ender-Grid.zip)

foreachthing, 2020-09-04:
I just downloaded your Ender-Grid_svg.zip and I must point out, that this is not a SVG as I know it.
Your SVG just imports an image! Here's what Inkscape had to say it:

[image]

And here's the content of your SVG:
```
<?xml version="1.0" encoding="utf-8"?>
<!-- Generator: Adobe Illustrator 16.0.0, SVG Export Plug-In . SVG Version: 6.00 Build 0)  -->
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg version="1.1" id="Layer_1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" x="0px" y="0px"
	 width="612px" height="792px" viewBox="0 0 612 792" enable-background="new 0 0 612 792" xml:space="preserve">
<image overflow="visible" width="2776" height="2717" xlink:href="Ender-Grid.png"  transform="matrix(1 0 0 1 -1082 -963)">
</image>
</svg>
```

I don't know Adobe Illustrator, but you need to export shapes (lines, circles, ...) not linking a PNG.

closed 2026-09-29 by prusaslicer-bot[bot]