prusa3d/PrusaSlicer #4690: Fails to compile after Merge branch 'tm_sl1_import_libpng' commit

Jebtrix, 2020-08-31:
### Version
Latest git master @ 143e3a6a3588fe467e3cb3940cf78df30af56f96

### Operating system type + version
Win7 64bit / VS2019 v14.27.29110 (latest)

### Behavior
Compiling fails with:
`Could NOT find PNG (missing: PNG_LIBRARY) (found version 1.6.35)`

Here is all CMakeOutput.log files and stdout/stderr from console during compilation:
[Failed-compile-logs.zip](https://github.com/prusa3d/PrusaSlicer/files/5152213/Failed-compile-logs.zip)

Jebtrix, 2020-08-31:
False alarm, updating CMake to latest 3.18.2 fixed the problem.

closed 2020-08-31 by Jebtrix

foreachthing, 2020-09-04:
Got the same issue, with cmake 3.18.2, though.
Any clues are appreciated!!

I was able to build PS for a long time - it just stopped working at some point ....

[CMakeError.log](https://github.com/prusa3d/PrusaSlicer/files/5173294/CMakeError.log)
[CMakeOutput.log](https://github.com/prusa3d/PrusaSlicer/files/5173295/CMakeOutput.log)

These are the messages from a rebuild (which has always worked):
_(Nevermind the path, I added a subdir to `c:\dev\` to start fresh. In the issue above I am in `c:\dev\1\`)_
```
       "C:\dev\PrusaSlicer\deps\build\ALL_BUILD.vcxproj" (default target) (1) ->
       "C:\dev\PrusaSlicer\deps\build\deps.vcxproj" (default target) (3) ->
       "C:\dev\PrusaSlicer\deps\build\dep_PNG.vcxproj" (default target) (6) ->
       (CustomBuild target) ->
         C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(234,5): error MSB6006: "cmd.exe" exited
       with code 1. [C:\dev\PrusaSlicer\deps\build\dep_PNG.vcxproj]
```

foreachthing, 2020-09-04:
### Tested commits:
#### working:
* 93921dc7c8d975f2155bfc9d60a7c78bf8bf2745

#### not working
* ba87a4fd9a607fd0ed7f60241a25884ff0d6d61b

I'll be testing and updating here.