prusa3d/PrusaSlicer #4691: Segmentation fault after install and first start

VaultPython, 2020-09-01:
### Version

|Information for package PrusaSlicer||
|-|-|
|Repository     | openSUSE-Tumbleweed-Oss|
|Name           | PrusaSlicer|
|Version        | 2.2.0-5.2|
|Arch           | x86_64|
|Vendor         | openSUSE|
|Installed Size | 17.4 MiB|
|Installed      | Yes|
|Status         | up-to-date|
|Source package | PrusaSlicer-2.2.0-5.2.src|

### Operating system type + version

openSUSE Tumbleweed
64-bit
GNOME Version: 3.36.5
Wayland

### 3D printer brand / version + firmware version (if known)

### Behavior

After the Installation via **sudo zypper install PrusaSlicer** the Programm won't start either via Gnome or Terminal command
The following Messages are displayed in the terminal:

```bash
flo@wlan-xxx-xxx-xxx-xxx:~> /usr/bin/prusa-slicer

(prusa-slicer:10446): GLib-GObject-WARNING **: 10:57:47.979: invalid (NULL) pointer instance

(prusa-slicer:10446): GLib-GObject-CRITICAL **: 10:57:47.979: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10446): GLib-GObject-WARNING **: 10:57:47.979: invalid (NULL) pointer instance

(prusa-slicer:10446): GLib-GObject-CRITICAL **: 10:57:47.979: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10446): GLib-GObject-WARNING **: 10:57:47.989: invalid (NULL) pointer instance

(prusa-slicer:10446): GLib-GObject-CRITICAL **: 10:57:47.989: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10446): GLib-GObject-WARNING **: 10:57:47.989: invalid (NULL) pointer instance

(prusa-slicer:10446): GLib-GObject-CRITICAL **: 10:57:47.989: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10446): GLib-GObject-WARNING **: 10:57:47.991: invalid (NULL) pointer instance

(prusa-slicer:10446): GLib-GObject-CRITICAL **: 10:57:47.991: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10446): GLib-GObject-WARNING **: 10:57:47.991: invalid (NULL) pointer instance

(prusa-slicer:10446): GLib-GObject-CRITICAL **: 10:57:47.991: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10446): GLib-GObject-WARNING **: 10:57:48.024: invalid (NULL) pointer instance

(prusa-slicer:10446): GLib-GObject-CRITICAL **: 10:57:48.024: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10446): GLib-GObject-WARNING **: 10:57:48.024: invalid (NULL) pointer instance

(prusa-slicer:10446): GLib-GObject-CRITICAL **: 10:57:48.024: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10446): GLib-GObject-WARNING **: 10:57:48.024: invalid (NULL) pointer instance

(prusa-slicer:10446): GLib-GObject-CRITICAL **: 10:57:48.024: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10446): GLib-GObject-WARNING **: 10:57:48.025: invalid (NULL) pointer instance

(prusa-slicer:10446): GLib-GObject-CRITICAL **: 10:57:48.025: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
Segmentation fault (core dumped)
flo@wlan-xxx-xxx-xxx-xxx:~> prusa-slicer

(prusa-slicer:10674): GLib-GObject-WARNING **: 11:02:25.151: invalid (NULL) pointer instance

(prusa-slicer:10674): GLib-GObject-CRITICAL **: 11:02:25.151: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10674): GLib-GObject-WARNING **: 11:02:25.151: invalid (NULL) pointer instance

(prusa-slicer:10674): GLib-GObject-CRITICAL **: 11:02:25.151: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10674): GLib-GObject-WARNING **: 11:02:25.160: invalid (NULL) pointer instance

(prusa-slicer:10674): GLib-GObject-CRITICAL **: 11:02:25.160: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10674): GLib-GObject-WARNING **: 11:02:25.160: invalid (NULL) pointer instance

(prusa-slicer:10674): GLib-GObject-CRITICAL **: 11:02:25.160: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10674): GLib-GObject-WARNING **: 11:02:25.163: invalid (NULL) pointer instance

(prusa-slicer:10674): GLib-GObject-CRITICAL **: 11:02:25.163: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10674): GLib-GObject-WARNING **: 11:02:25.163: invalid (NULL) pointer instance

(prusa-slicer:10674): GLib-GObject-CRITICAL **: 11:02:25.163: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10674): GLib-GObject-WARNING **: 11:02:25.196: invalid (NULL) pointer instance

(prusa-slicer:10674): GLib-GObject-CRITICAL **: 11:02:25.196: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10674): GLib-GObject-WARNING **: 11:02:25.196: invalid (NULL) pointer instance

(prusa-slicer:10674): GLib-GObject-CRITICAL **: 11:02:25.196: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10674): GLib-GObject-WARNING **: 11:02:25.196: invalid (NULL) pointer instance

(prusa-slicer:10674): GLib-GObject-CRITICAL **: 11:02:25.196: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:10674): GLib-GObject-WARNING **: 11:02:25.196: invalid (NULL) pointer instance

(prusa-slicer:10674): GLib-GObject-CRITICAL **: 11:02:25.196: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
Segmentation fault (core dumped)
```

How can i resolve this issue

lukasmatena, 2020-09-01:
I cannot tell for certain without seeing a backtrace, but this very much resembles https://github.com/prusa3d/PrusaSlicer/issues/2515. The problem in that case was linking against wxWidgets 3.0 and GTK3, which apparently does not work.

I looked up the package in [OpenSUSE repository](https://build.opensuse.org/package/show/openSUSE:Factory/PrusaSlicer) and it seems to depend on wxGTK3-devel package. Our builds are build and tested againts wx3.1 and GTK2. Other combinations are untested and discouraged, although a hard crash may be a bit over the edge.

VaultPython, 2020-09-01:
Hello @lukasmatena and thank you for your reply,
is there anything i can do to resolve this issue or any hints that could help me i am relatively new to Linux.

lukasmatena, 2020-09-01:
Our goal is to migrate to GTK3 completely, but it will take some effort. Until then we cannot do much more than notify package maintainers. I believe it would be @bnavigator in this case. Any thoughts?

You can use our [in-house bred AppImage](https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.2.0/PrusaSlicer-2.2.0+linux-x64-202003211856.AppImage) in the meantime. That will presumably not crash (I hope).

bnavigator, 2020-09-01:
Thank you for the ping, I will look into this.

The build against GTK3 has been requested specifically to support Wayland.
https://build.opensuse.org/request/show/825277
Ping @StefanBruens

bnavigator, 2020-09-01:
@VaultPython could you please try to start PrusaSlicer with
`GDK_BACKEND=x11 prusa-slicer` and report whether this works for you?

If that does not help, until we resolved this issue, you can use the rpm from [here](https://build.opensuse.org/package/binaries/home:bnavigator:prusa-gdk2/PrusaSlicer/openSUSE_Tumbleweed) as soon as the build is finished (Nothing there yet, the build takes a few minutes).

VaultPython, 2020-09-01:
Hello @bnavigator the command **GDK_BACKEND=x11 prusa-slicer** i can now use the PrusaSlicer but the terminal still puts the following text out:
```bash
flo@wlan-xxx-xxx-xxx-xxx:~> GDK_BACKEND=x11 prusa-slicer

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:27.933: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:27.933: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:27.933: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:27.933: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:27.942: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:27.942: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:27.942: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:27.942: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:27.944: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:27.944: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:27.944: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:27.944: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:27.970: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:27.970: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:27.970: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:27.970: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:27.970: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:27.970: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:27.970: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:27.970: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:28.123: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:28.123: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:28.123: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:28.123: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:28.133: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:28.134: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:28.134: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:28.134: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:28.137: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:28.137: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:28.137: invalid (NULL) pointer instance

(prusa-slicer:12528): GLib-GObject-CRITICAL **: 13:23:28.137: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:12528): GLib-GObject-WARNING **: 13:23:28.140: invalid (NULL) pointer instance

[... 648 lines cut ...]
(prusa-slicer:12528): Gtk-CRITICAL **: 13:25:15.477: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:12528): Gtk-CRITICAL **: 13:25:15.477: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:12528): Gtk-CRITICAL **: 13:25:15.477: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:12528): Gtk-CRITICAL **: 13:25:15.478: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:12528): Gtk-CRITICAL **: 13:25:15.567: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:12528): Gtk-CRITICAL **: 13:25:15.568: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:12528): Gtk-CRITICAL **: 13:25:15.568: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:12528): Gtk-CRITICAL **: 13:25:15.570: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:12528): Gtk-CRITICAL **: 13:25:15.570: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:12528): Gtk-CRITICAL **: 13:25:15.570: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:12528): Gtk-CRITICAL **: 13:25:15.570: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:12528): Gtk-CRITICAL **: 13:25:15.571: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
```
I sliced a .stl file to test it and it worked, thank you for your help.

StefanBruens, 2020-09-01:
It works fine with Gtk3 and X11.

Fedora wants to remove Gtk2 for F34, Debian is also trying to get rid of it, and on openSUSE TW IIRC the last major package with a Gtk2 dep is Gimp (Gtk3 version to be released soon™).

So Gtk2 is a dead horse ...

bnavigator, 2020-09-01:
The same Gtk-CRITICAL messages appear on Gtk3 and X11, the Glib-Gobject warnings even with Gtk2. @StefanBruens can we ignore them?

StefanBruens, 2020-09-01:
I don't think I have ever seen a nontrivial GTK app which starts without warnings, so yes (unless GTK maintainers decide to make these errors fatal some day).

bnavigator, 2020-09-01:
The next update of the package should resolve this.
https://build.opensuse.org/request/show/831122

lukasmatena, 2020-09-01:
@StefanBruens @bnavigator 
Thanks, guys.

Just for possible future readers, I understand that you link against the same package (GTK3), but enforce fallback to Xorg by the env variable `GDK_BACKEND` which resolves the crash on Wayland.

> I don't think I have ever seen a nontrivial GTK app which starts without warnings

We feel less ashamed now. Cheers for that.

I'm inclined to leaving this open so it is easily discoverable when we do the GTK3 migration.

bnavigator, 2020-09-01:
In an unfortunate coincident, the boost package for Tumleweed got updated to 1.74 today. This triggered #4264. I am applying #4340, but it means that the new PrusaSlicer package will be a bit delayed.

bubnikv, 2020-10-09:
We are now setting GDK_BACKEND environment to "x11" at the start of PrusaSlicer on Linux, which should fix this issue.
The change will be in PrusaSlicer 2.3.0-alpha1.
Closing.

closed 2020-10-09 by bubnikv