prusa3d/PrusaSlicer #4694: Advanced Math Syntax, Formula Documentation

eduncan911, 2020-09-01:
### Version
2.2.0

### Operating system type + version
Linux (Pop_OS 20.04, aka Ubuntu 20.04)

### 3D printer brand / version + firmware version (if known)
Doesn't matter (I have many brands and versions)

### Behavior
Unable to write advanced math formulas in Start Gcode and other gcode locations.

* There doesn't seem to be any documentation on the official PrusaSlicer website, nor here in this github `docs` folder, on the syntax for writing formulas, math, and what advanced functions there is available.
* Using Slic3r functions, which I thought PrusaSlicer was based on, does not work .

Slic3r says I should be able to use `{{` to switch to math, such as logarithmic formulas and cosign.  However, PrusaSlicer errors on all Slic3r examples of math, stating:

```
Parsing error at line 19. Unknown syntax error
M221 S{{ 100 - ( layer_z log 1.2 ) }}
      ^
```

If I remove a curly bracket, i then get errors such as:

```
Parsing error at line 19. Expecting tag literal-char
M221 S{ 100 - ( layer_z log 1.2 ) }
                        ^
```

Switching things around, I tried to use `log` like `min/max`:

```
Parsing error at line 19: Not a variable name
M221 S{ 100 - log( layer_z, 1.2 ) }
              ^
```

Am I missing a document somewhere?

neophyl, 2020-09-01:
I take it you looked at the wiki page here ? https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language

eduncan911, 2020-09-01:
"Macro", so that's the term I should have searched for.  :)

No, I haven't seen that page.  However, it shows the limitations of Slic3r/PS in using log function.

```
Allowed functions
min(a, b), max(a, b)
int(a)
```

Guess I'm going to have to go the Klipper route and write my own functions.

closed 2020-09-01 by eduncan911