prusa3d/PrusaSlicer #4695: Feature Request: Check for non-ASCII characters during (USB-)export

MartinMajewski, 2020-09-02:
This is a follow-up feature request related to the bug discussed here:

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/791

and

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/757

As the (current) firmware is not able to handle non-ASCII characters in the filenames, resulting in a crash, I want to propose the following feature in PrusaSlicer:

During export, PS could warn the user that the filename contains non-ASCII characters and provide three options:

1. Fix the filename automatically: The non-ASCII character gets either converted into it's ASCII representation or if this representation could also cause issues (Unicode ä = &#xE4;), replace the Unicode character with a placeholder like an underscore. I think providing just one of both possibilities would be enough.

2. Cancel export and edit manually: Let the user return to the export dialog and change the filename to something that fits.

3. Continue anyway: The blame is on the user.

One should be able to deactivate this check via the preferences menu.

I am not fully aware if this is just an FW bug (a missing feature, respectively), or if it is a limitation to the FAT32 format. So I cannot tell if there is a sufficient fix possible (except for not having the FW crash). However, I think it would be an excellent addition to PrusaSlicer to ensure a more uniform filename scheme for the GCode files. Also, this feature would prevent crashes on older / other printer-platforms.

Cheers,
Martin

neophyl, 2020-09-02:
Also useful for those using flash air cards as if you use a non allowed character on those it will copy to the card but then that directory will be inaccessible over wifi and you need to put it into the computer to delete the offending file.

jokohoko, 2020-09-04:
See also: 
https://github.com/prusa3d/Prusa-Firmware/issues/2130
and
https://github.com/prusa3d/PrusaSlicer/issues/2800

They solved it back then with a firmware update on the MK firmware.

MartinMajewski, 2020-09-04:
> See also: 
> 
> https://github.com/prusa3d/Prusa-Firmware/issues/2130
> 
> and
> 
> https://github.com/prusa3d/PrusaSlicer/issues/2800
> 
> 
> 
> They solved it back then with a firmware update on the MK firmware. 

Irrelevant! PrusaSlicer is not only for Prusa-Printers. And the Mini, e.g., still suffers from that. 

This feature would add another layer of safety and reliability.

jokohoko, 2020-09-04:
I agree with you @MartinMajewski! I showed these issues to remind them that this was an issue before. And that back then it was also suggested to make it safe for all situations (that's why i made both a firmware and software issue).

closed 2026-09-29 by prusaslicer-bot[bot]