prusa3d/PrusaSlicer #4696: First layer height with different type of nozzles

Ondrashek, 2020-09-02:
### Version
Version 2.2.0+linux-x64
OS: Linux Mint
Original Prusa Mini, FW 4.0.5

### Behavior

In PrusaSlicer with different type of nozzles the first layer height is still 0.2. No matter, if I choose in print settings 0,3 QUALITY, 0,4 DRAFT etc. 
I need to manually set up different first layer height similar to layer height for example: layer height 0,3 = first layer height 0,27.

rtyr, 2020-09-02:
Yes, it is intentional and used on all our printers for a very long time. Your underextrusion issues are most likely caused by something else (drifting live z?).

Ondrashek, 2020-09-02:
> Yes, it is intentional and used on all our printers for a very long time. Your underextrusion issues are most likely caused by something else (drifting live z?).

No, my underextrusion cause, because I forget change first layer height with different layer height. For 3D printing novice would be nice, if first layer height can be calculated automatically.

rtyr, 2020-09-02:
Like I said, 0.2 is intentional. This way, first layer is consistent for all profiles. There is no need to use same layer height as rest of the print, you can use lower (up to 0.15 in case of Prusa printers or higher)...

If you for some reason want same first layer height as rest of the print, set value to 100%.

Please provide your problematic 3MF project file, I can check it.

rtyr, 2020-09-16:
No reaction. Closing.

closed 2020-09-16 by rtyr