prusa3d/PrusaSlicer #4697: [Feature request] Green and red flags for filaments in MMU multi-mode in <Platter> screen

LesterHG, 2020-09-02:
New feature request
### Version
Slicer 2.2.0
### 3D printer brand
Original Prusa i3 Mk3S MMU2S

**Problem**:  Slicer only permits system preset filaments for MMU extruders in multi-mode.

**Problem example**:  I want to print a model using PLA and Semiflex 98A in each of two extruders in MMU multi-mode.  I cannot assign the Semiflex 98A preset to an extruder because it is not a system preset.

**Work-around**:  Load an existing system preset, for example, PrimaSelect PVA.  Change every value in every section to be equal to the values required as listed in the non-system preset for SemiFlex 98A.  Save the modified preset as “Was PVA now 98A”.  Can now select the user preset “Was PVA now 98A” and assign it to any extruder in multi-mode.

**Work-around success**:  I have been able to print the part in the way I want, with few problems due to swapping between the PLA and 98A filaments.  Of the 76 tool changes for my print, only 1 failed, and I was able to correct that manually at the time and finished the print successfully.

**Feature request**:  Provide green and red flag lists of presets for extruders in <Platter> when MMU is selected as the printer (as implemented for <Print Settings> and <Filament Settings>).  Green flag restricts filament assignments to system presets as currently.  Red flag permits assignment of any preset.

**Reasoning**:  
- The work-around is straightforward, but time-consuming.  
- In my case, the lack of this feature wasted hours of the Prusa support team's time while they attempted to explain to me why I should not be permitted to load any filament into any extruder.  
- When I refused to accept that explanation, it then wasted hours of the Prusa support team's time while they dribbled various hints to me which made no sense at the time until it finally dawned on me how to do what they were not allowed to tell me to do.  
- It also wasted hours of my time.
- It made me think I had made a stupid mistake in purchasing the MMU.
- The required filament settings are already fully setup in existing presets. There is no good reason to prevent their use.  Red flag them if that is thought desirable.

rtyr, 2020-09-02:
All filament profiles, which are not marked "MMU2" have MMU1 specific toolchange parameters (=completely different). It means that those filament profiles would simply not work on MMU2 without adjusting most of the toolchange parameters. Your work-around was actually correct and faster than reconfiguring all specific mmu parameters of existing profile.

LesterHG, 2020-09-02:
The Prusa-supplied "Semiflex or Flexfill 98A" profile works fine with my MMU2S in single-mode.  Therefore I'm thinking that its parameters are indeed MMU2-adjusted (as illustrated in the screenshot attached).  If it works in single-mode, why would it not work in MMU2S multi-mode?

[image]

rtyr, 2020-09-02:
No, you can see on your screenshot, that toolchange parameters are not for MMU2. There are no toolchanges in single mode.

LesterHG, 2020-09-02:
Sorry, I cannot see that.  There is no mention of MMU2 (or MMU1) in the screenshot.  There is mention of MM, and, strictly speaking, the MMU1 or MMU2 is a MM of multiple filament feeders connected to just the single extruder toolhead.  (The terminology confusion in Slicer is sad to see, and perhaps contributes to my incomprehension.)

So that I may understand correctly, are you saying that the listing of toolchange parameters that I can see in the screenshot are parameters suited only for the MMU1 and are unsuited for the MMU2?  And, they are parameters which do not relate to multi-mode use of the filament but relate only to single-mode use?

rtyr, 2020-09-02:
> So that I may understand correctly, are you saying that the listing of toolchange parameters that I can see in the screenshot are parameters suited only for the MMU1 and are unsuited for the MMU2? 

Yes.

> And, they are parameters which do not relate to multi-mode use of the filament but relate only to single-mode use?

Toolchange parameters are used only for multi-material prints (not important for single-mode use). That is the reason why common filament profiles are not restricted in single-mode. Profiles for MMU2 (MM mode) are completely separate profiles with different settings (temperatures, ramming, toolchange parameters).

LesterHG, 2020-09-02:
So, again just to be clear, the toolchange parameters listed in the screenshot *are* multi-mode parameters?  

I am not talking about profiles here, just parameters.  I understand that a profile is a set of the values, or a list of the settings, of the parameters.  I understand you to say that the *values* of the toolchange parameters listed in the screenshot, that is, the settings of the parameters, relate to the MMU1.  

But the parameters themselves -- loading and unloading speeds, loading and unloading times, delays and moves -- these are parameters which relate to multi-mode use, surely?

rtyr, 2020-09-02:
Toolchange parameters are used for multimaterial prints (with toolchanges, wipe tower, ramming, ...).

LesterHG, 2020-09-02:
OK, good progress, thanks. I'm trying to be careful with terminology, already a slight misunderstanding about parameters and values of parameters, my apologies.  But instead of multimaterial prints can we please talk about "multi-mode"?  Single-mode can also yield multimaterial prints if applied sequentially....

I am not sure that the toolchange parameter values are only for the MMU1.  I say that because I specifically used the toolchange values from the 98A profile, shown in my screen shot, to print parts in multi-mode perfectly successfully on the MMU2S.  I'm not sure how that would happen if the parameter values were systematically unsuited to the MMU2.  In particular, I was careful to copy the ramming values, and they worked.  I mentioned that I had only one one failure earlier, and it was in fact the loading of the PLA filament which failed, not the loading of the semiflex 98A; that did surprise me.

So this brings us back to the point of my suggestion for a new feature.  In due course, sooner or later, there will be a variety of profiles which apply to the MMU2 multi-mode.  Already, there is a variety which apply to MMU2 single-mode, and these are accessed through green and red flags.  It seems sensible to me that similar green and red flag access should be given to the filament profiles for the extruders in the <Platter> screen for multi-mode.  I don't think the suggestion is invalid simply because there happen to be no profiles with "@MMU2" currently listed.

On the point about the toolchange parameters not being used for single-mode, where are these different single-mode parameters found, please?  Some of the parameters listed -- loading and unloading speeds, loading and unloading times, for example -- seem just as relevant for single-mode use.  The toolhead must be loaded with the filament, after all, and unloaded after use.  It is not obvious to me why the parameters, or indeed the parameter values, should be different between single-mode and multi-mode.

rtyr, 2020-09-02:
> I am not sure that the toolchange parameter values are only for the MMU1. I say that because I specifically used the toolchange values from the 98A profile, shown in my screen shot, to print parts [...]
Yes, they are only for the MMU1 and they shouldn't be used on MMU2. MMU1 is completely different hardware (it is bowden printer with 4 separate extruders). Yes, ramming sequence can work, even if it is not optimized/tested with MMU2, but most of other parameters not. I believe there is no need to discuss this further. Of course, users can make their own custom profiles (like you did, based on PVA) but such profile should be based on some system "MMU2" filament profile. Users can experiment with the ramming sequence and cooling moves to improve realiability with their specific filament, it is expected. 

> So this brings us back to the point of my suggestion for a new feature. In due course, sooner or later, there will be a variety of profiles which apply to the MMU2 multi-mode. Already, there is a variety which apply to MMU2 single-mode, and these are accessed through green and red flags. It seems sensible to me that similar green and red flag access should be given to the filament profiles for the extruders in the screen for multi-mode. I don't think the suggestion is invalid simply because there happen to be no profiles with "@MMU2" currently listed.
> 
I don't agree. We don't want incompatible system filament profiles to be used for multimaterial printing (multi-mode) for the reasons I mentioned. 

> On the point about the toolchange parameters not being used for single-mode, where are these different single-mode parameters found, please? Some of the parameters listed -- loading and unloading [...]

Because actions needed for single-mode print are handled completely by printer firmware. Those actions are initiated by Tx,Tc commands in the start g-code and "M702 C" in the end g-code.

rtyr, 2020-11-16:
I believe we can close this.

closed 2020-11-16 by rtyr