prusa3d/PrusaSlicer #4702: Feature Request : Split Wipe Tower

weckso, 2020-09-03:
Would like the option to have independent wipe towers per material instead of all the material using the same block. Use case is for an IDEX configuration having the left extruder use a wipe tower near the far left location and the right extruder tower far right.

weckso, 2020-09-05:
Another use case would be something like Polymakers Polysupport which has minimal adhesion with certain materials. The single block just ends up becoming a giant mess with bigger prints.

Kachidoki2807, 2020-12-14:
I have also the issue with the PolySupport filament.

IdeaMaker have multiple options regarding the wipe tower. It can interlace the material like the PrusaSlicer tower, but it can also nest the two materials into the tower (side-by-side, not stacked) or each material can have its own tower. Unfortunately the both tower are undissociable and the placement is not as easy as in PrusaSlicer. 

Another interesting feature is the ability to hollow the tower. On IDEX printer the purge volume can be drastically reduced, that introduce two issues:
- First is the default minimal purge volume into the filament profiles (15mm³), and the default purging volume of the wipe tower (140mm³) that are way too large for an IDEX. Based on my tests, 5mm³ is more than enough for PLA, maybe some tricky filaments (like PolySupport) needs a bit more.
- The second problem is the footprint that become too small to ensure a good stability of the tower on tall prints.
Hollowing the tower allow to increase the footprint without increasing the material usage.

I hope PrusaSlicer's devs will improve the IDEX management in a near future. I figured out that there is no common slicers on the market that fully properly handle IDEX printers, even with a quiclky growing chinese IDEX printers range. But there is not much missing to PrusaSlicer to get a perfect outcome.

rtyr, 2021-02-24:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3451 and https://github.com/prusa3d/PrusaSlicer/issues/3070. Closing this one.

closed 2021-02-24 by rtyr