prusa3d/PrusaSlicer #4704: ender 3 pro

tree124, 2020-09-04:
i got an ender 3 pro for Christmas for some reason that prints that i have been getting of the creality slicer are not flat so i decided to try Prussia slicer but for some reason the filament is not sticking to the bed at all but then got back to the creailty one and it sticks to the bed fine i dont understand what the problem is

my temps are 215 hot end and 70 on the bed

jokohoko, 2020-09-04:
First layer extrusion width or first layer height settings might be different.
Also the Z-offset settings is sometimes set by default on some slicers. With that you can reduce or raise the first layer in reference to the endstop.

rtyr, 2020-09-04:
As jokohoko pointed out, your first layer setting are most likely different.

rtyr, 2020-09-09:
Closing. Feel free to reopen if you find some related bug in the PrusaSlicer.

closed 2020-09-09 by rtyr