prusa3d/PrusaSlicer #4706: Feature request: speed modier blocks

neroes, 2020-09-04:
hi i print alot of miniatures, and sometimes there is alot of thin structures, so i would like to request the option to put up blocks that reduce the speed when inside a block to reduce the chance of accidentally tipping something through momentum

e.g. [this spider](https://forum.prusaprinters.org/wp-content/uploads/2020/09/spiderimage.png), the legs sometimes tip over, so it would be nice if i could turn down the speed around them

also thanks for the great work guys

rtyr, 2020-09-04:
It is already implemented.

See https://help.prusa3d.com/en/article/modifiers_1767.  

[image]

neroes, 2020-09-05:
oh sorry about that, hadn't been able to find it

thanks though

closed 2020-09-05 by neroes