prusa3d/PrusaSlicer #4707: Duration of Parking-CoolDown moves not adjustable (Multimaterial extruders)

taykofox, 2020-09-04:
### Version
PrusaSlicer-2.2.0+linux-x64-202003211856.AppImage

### Operating system type + version
Ubuntu 20.04 LTS (focal fossa)

### 3D printer brand / version + firmware version (if known)
- modded Monoprice Select mini 
- SKR v1.3
- marlin 2.0.9
- BTT zsyong Dual-Extruder (2in1out) 
- Everything (finally ;-) is set up correctly, it works fine! 

### Behavior
- It takes very long to purge and change the filament (extruder). I would like to lower the time that is used to cool the old filament down before parking it. Right now its 15 seconds (per layer!). I really think 5 seconds would be enough (at least for a zsyong, since it stays remarkably cold!). 

### Proposal
It would be perfection to have a number-selector to change "Time to cool down" in the "Single Extruder MM setup"-Page. Either lower the number cooling moves or the speed of them. 
**_Is this a new feature request?_ YES**
I attached a gimped sketch how it could look like ;-)
[image]

#### Project File (.3MF) where problem occurs
n./a.

**Thank you so much for PrusaSlicer! Its a great tool !!! Will really appreciate this (new) feature. If I can help, just ask, will gladly do what I can to support this!**

rtyr, 2020-09-04:
Filament Settings -> Advanced

[image]

taykofox, 2020-09-04:
Thank you very much! Thats like way more than I would expect, but I surely will try millions of combinations :-D
Problem solved. AWESOME community!
Thanx again!

closed 2020-09-04 by taykofox