prusa3d/PrusaSlicer #4708: Move/override Elephant foot compensation

blddk, 2020-09-05:
### Version
2.2.0

### Behavior
I got multiple printers, and the elephant foot is a bit different on each. But the elephant foot compensation is set in "Print Settings", so I can't adjust it for each printer, without making a set of print settings for each printer.

_Is this a new feature request?_
Yes/no, either having an override in the Printer Settings tab, or moving the compensation to the Printer Settings tab, will allow a printer depending variable to be in the Printer Settings settings.

Kadah, 2020-09-13:
This same logic would apply to all settings too. Print settings should likely be unique to each model of printer for that reason.

I use the dependencies to control what printers my custom profiles are valid for.

blddk, 2020-09-13:
> This same logic would apply to all settings too. Print settings should likely be unique to each model of printer for that reason.

How so? I see layer height, infill and such as generic settings for the model I want to print, but elephant foot got nothing to do with the model itself, it is the printer making it.

closed 2026-09-29 by prusaslicer-bot[bot]