prusa3d/PrusaSlicer #4709: Bottom layer cannot change color in preview

ArdjanB, 2020-09-05:
### Version
Version 2.2.0+win64

### Operating system type + version
Windows 10, v2004, Build 19041.450

### 3D printer brand / version + firmware version (if known)
MK3s (upgraded from purchased MK3. Firmware 3.9.0-3421

### Behavior
in Prusaslicer, when I want to change the preview colors when doing a multicolor print (=rightclick on portion of layer bar to change the color), I cannot change the color of the botton color. When wanting to do a preview a print using the 7-colors of the pride flag, I can change the upper 6 colors, but not the bottom color. See pictures: the arrows point to where I want to right-click.

[image]
[image]

#### Project File (.3MF) where problem occurs
No loaded 3mf files: just dragged some STL in the slicer. Restarted the slicer, same result with different STLs.

rtyr, 2020-09-05:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4309. Color of the bottom part (before any color change) is taken from filament settings. You can easily change it by clicking on the left icon (see attached image). Closing.

[image]

closed 2020-09-05 by rtyr

brettvitaz, 2020-09-24:
This is brought up a lot. Why not make it work consistently across all layers? Changing color at the lowest layer should accomplish the same thing as clicking the color swatch of the default filament.