prusa3d/PrusaSlicer #4711: Ctrl+I shortcut overridden

wavexx, 2020-09-07:
### Version
573194e059836916b6f216dc068c27a89ea7b843

### Operating system type + version
Debian unstable / X11

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Ctrl+I was (and still is, according to menus and help) assigned to "Import STL/OBJ..".
Now pressing Ctrl+I opens a new instance (visible in the Window menu).

* _Expected Results_

"New instance" should probably be assigned to a different shortcut.

_Is this a new feature request?_
No

stefanct, 2020-10-20:
The whole instance handling seems a bit weird to me. I would have expected ctrl+n to open a new instance and certainly not reset the current one (thus I would rather see the the whole topic dealt with differently). In any case I can confirm this issue with current master (and came here to report it originally). I have cherry-picked the patch (on top of a61d0a953) and can confirm that it works fine.

bubnikv, 2020-10-20:
> I would have expected ctrl+n to open a new instance

Fair enough, collisions of keyboard shortcuts is not a good thing. We will
check whether Ctrl-N does not collide with another shortcuts and that it is
passable on all platforms.

> and certainly not reset the current one

??

> (thus I would rather see the the whole topic dealt with differently).

We are open to suggestions.

wavexx, 2020-10-20:
Is opening a new instance through the menu faster than starting a new process, or is the single-instance management needed by MacOS? It's a genuine question, since I never enabled "since instance", and usually just start multiple processes. Start time with either feels mostly on-par when using linux.

bubnikv, 2020-10-20:
> Is opening a new instance through the menu faster than starting a new
process

No, it is the same.

>  or is the single-instance management needed by MacOS? ?

Yes.

Once we had it implemented for OSX, we thought it may be useful on other
platforms as well.

FidelCapo, 2020-10-20:
@stefanct Unfortunately, the shortcut ctrl-N is already used for "New Project"..

stefanct, 2020-10-21:
Apparently I did not make clear what I meant, sorry.

A common user interface behavior is that ctrl+n creates new files/windows, either as new windows or tabs/subwindows. The previously focused entity remains untouched and one can continue to work on that or close it independently. This is pretty much the expected behavior since windows 3.11(?) as well as most (all?) linux desktop environments. No idea about the golden cage followers, but I'd be surprised if it's much different there.

In current slicer this is not the case. Instead, when executing ctrl+n or file/new project the current project gets reset/lost. If a user wants to accomplish what I have described in the previous paragraph instead, the user needs to find the "Open new instance" entry in the window menu or use the ctrl+i (or ctrl+shit+i) shortcut originally discussed in this issue.

If I would have to decide on this issue then I would change the existing ctrl+i shortcut to ctrl+n and move the "new instance" entry to file/new project.
The existing file/ entry that resets the project to its initial state would be renamed. I am not entirely sure what the best wording would be - I can't think of any application that has a similar feature. Usually they just have the traditional "new" method I keep rambling about but sometimes there is a way to revert to the previously saved state (i.e. the respective file is re-read from disk and the state of the application instance is reverted to reflect that). The latter is often called revert or reload from disk.

This is not super important but it is quite perplexing for new users (the shortcut collision didn't help me there either :).

Haprog, 2020-11-11:
I just tested this on alpha3 (64bit, Windows 10).

Ctrl+I seems to be mapped both for `File / Import / Import STL/OBJ/AMF/3MF...` and also for `Window / Open new instance` (at least that is displayed in the menus) but actually pressing Ctrl+I seems to always trigger the import dialog for me.

Ctrl+N seems to be mapped only to `File / New Project` and works like it used to (it clears the current instance by triggering at least "Delete all"). Though I saw at least one guy in a Facebook group [complain](https://www.facebook.com/groups/Prusai3users/permalink/1658902024294272/?comment_id=1658924390958702&__cft__[0]=AZV1sPYEJBGazUAROKnuwYj73_wmog9xdB89AjT_H3jDRgl-Fp54NmvZCirVv8orR2JWE2Fz8JH9sD5dh7fKgorH-HDyj_66_amPEoEZTqTlrt-OqldQzITrnPOdxQoQ8L4&__tn__=R]-R) about Ctrl+N triggering new instance for him instead of "New Project", which he didn't like, since he's used to using that shortcut to clear the current instance instead of opening a new one. I think backwards compatibility and habits should be considered also when changing old shortcuts.

Haprog, 2020-11-12:
I just tested this on Linux too and I can verify that Ctrl+I on Linux triggers the "Open new instance" but on Windows 10 it triggers the import dialog.

I've tested the versions:
- PrusaSlicer-2.3.0-alpha3+win64-202011041552.zip (on Windows 10)
- PrusaSlicer-2.3.0-alpha3+linux-x64-202011041552.tar.bz2 (on Pop!_OS 20.10, Ubuntu based)

nordurljosahvida, 2020-12-06:
Same thing here on Ubuntu 20, I can no longer import objects via ctrl-I, as that opens up a new instance.

PR #4734 (2020-12-07): Use Ctrl+Shift+I to open a new instance instead of Ctrl+I
"Open new instance" overrides the "import" shortcut.

Use Ctrl+Shift+I for new instance, which is definitely less important
than importing new models.

Also document the shortcut in the Keyboard shortcuts dialog.

Fixes #4711
commit 1fd54938 (2020-12-07): Use Ctrl+Shift+I to open a new instance instead of Ctrl+I

"Open new instance" overrider the "import" shortcut.

Use Ctrl+Shift+I for new instance, which is definitely less important
than importing new models.

Also document the shortcut in the Keyboard shortcuts dialog.

Fixes #4711
closed 2020-12-07 by bubnikv