prusa3d/PrusaSlicer #4713: [Feature Request] allow cooling fan speed values to be set per object/part

jhoblitt, 2020-09-07:
When printing a platter with multiple mechanical parts in PETG, I would like to be able to mix in some small / fine details parts that need to be printed with 100% fan speed.  As of PrusaSlicer 2.2.0, there is no mechanism to set per object/part cooling values.

treyd3, 2021-02-10:
I second this.  
Currently I have to go and modify the filament profile for every print if my default setting aren't adequate and as stated above, that change applies to the entire plate. 
Please add part cooling settings to modifier and per-part menus. 
In fact, I don't really understand why every setting isn't available to modifiers and per-part menus. 

thanks!

rtyr, 2021-02-24:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4461. Closing.

closed 2021-02-24 by rtyr