prusa3d/PrusaSlicer #4715: M106 Added by Slicer by mistake! - Repeatable

Vlaka456, 2020-09-08:
### Version
2.2.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
SKR PRO 1.2

### Behaviour
I have been trying to print with PETG but the prints stop due to thermal runaway.  I tried reslicing.  The prints stop at the same instant.  I found the reason.

I have both of the cooling fan options unchecked on the slicer profile but the gcode has fan commands in it!  No reason.  The print is flat.  No overhangs.  The layer sizes are huge.  There is NO reason for these fan commands.  I searched the GCODE and found multiple M106 and M107 commands.

The fan triggers at Z2.2 on a 0,2 layer print... even though the print is consistent layer to layer pretty much.  Here is the first reference to M106:

G1 X134.629 Y1.543 E0.04645
G1 X134.931 Y1.567 E0.05712
G1 X136.683 Y2.152 F4800.000
G1 F1200.000
G1 X136.815 Y2.334 E0.06234
G1 X136.948 Y2.516 E0.06836
G1 X137.080 Y2.698 E0.07519
G1 X137.213 Y2.881 E0.08282
G1 X137.073 Y3.880 F4800.000
M106 S255
G1 F3600.000
G1 X137.920 Y4.727 E0.14771
M107
G92 E0
G1 X144.929 Y10.443 F4800.000
M106 S255
G1 F3600.000
G1 X137.536 Y3.050 E0.56632
G1 X137.187 Y3.303 E0.58965
G1 X137.196 Y3.356 E0.59258
G1 X144.727 Y10.887 E1.16943
G1 X144.727 Y11.534 E1.20446
G1 X141.236 Y8.043 E1.47184
G1 X141.867 Y9.281 E1.54709
G1 X141.875 Y9.328 E1.54968
G1 X144.727 Y12.181 E1.76816
G1 X144.727 Y12.827 E1.80319

Other than the fan check boxes is there another location that would trigger a fan?

Vlaka456, 2020-09-08:
I found the source of the unneeded fan codes.  It looks like the "Bridging" fan setting adds code even when the two fan checkboxes at the top are unchecked.  The areas where the fan was triggering had absolutely no bridging, the entire print had zero bridging.

Setting the bridging fan speed to 0% got rid of the gcode entries.

rtyr, 2020-09-08:
>  It looks like the "Bridging" fan setting adds code even when the two fan checkboxes at the top are unchecked.

Intended behavior.

>  The areas where the fan was triggering had absolutely no bridging.

Maybe you have "Detect bridging perimeters" parameter enabled? It is not possible to evaluate it unless you provide your 3MF project file.

rtyr, 2020-09-20:
No response. Closing.

closed 2020-09-20 by rtyr

Vlaka456, 2020-09-20:
I didnt notice the reply.  I had a weird bridging condition.  There were
automatic supports but there was a small section that didnt have supports.
This triggered the bridging fan.  I didn't think the fan itself was enough
to thermal runaway detect fail the hotend... but even with a silicone sock
it did.  One abnormal thing I was doing is I had the temps cranked to 260C
so the part cooling fan maybe had more effect as it pulled the temp down to
250C and held it there.

Perhaps there could be a disable all cooling fan checkbox at the top?