prusa3d/PrusaSlicer #4716: boost::filesystem::copy_file: The system cannot find the file specified:

TurboCooler, 2020-09-08:
### Version
_Version of PrusaSlicer used goes here_
Version 2.2.0+win64

_Use `About->About PrusaSlicer` for release versions_
Version 2.2.0+win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Microsoft Windows 10 Pro Version	10.0.18363 Build 18363

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa MK3S Firmware 3.9.0

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

Go to Configuration --> Check for Updates

Returns this error
boost::filesystem::copy_file: The system cannot find the file specified: 
"C:\Users\myusername\AppData\Roaming\PrusaSlicer\cache\PrusaResearch.idx-update", 
"C:\Users\myusername\AppData\Roaming\PrusaSlicer\vendor\PrusaResearch.idx-update"

[image]

_Is this a new feature request?_
No

Is there a way to work around this issue?

bgiot, 2020-09-21:
I have the same issue on MacOS... same PrusaSlicer version: 2.2.0

VictoryRice, 2020-09-23:
I have exactly the same issue as the OP

jmichael16, 2020-10-03:
Same here! How will I check for updates now?

trondkla, 2020-11-26:
Same here, windows 10 64 bits, Verson 10.0.18363 Build 18363, PrusaSlicer-2.2.0+win64-202003211145

kretzu77, 2020-12-09:
Run Prusa "As Administrator" (even you are administrator)

bubnikv, 2020-12-09:
Would you please re-test with https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-beta2 ?
We don't know what is going wrong, however we improved error reporting of errors when copying a file.
Thank you.

TurboCooler, 2020-12-09:
> Run Prusa "As Administrator" (even you are administrator)

If I run as "As Administrator" I do not get the error.  I get a message which for me makes more sense:

[image]

TurboCooler, 2020-12-09:
> Would you please re-test with https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-beta2 ?
> We don't know what is going wrong, however we improved error reporting of errors when copying a file.
> Thank you.

I am working but I will try the Beta this evening and see what is does.

kocikdav, 2020-12-11:
Please try also new beta (3) when its released. There will be improvements in error reporting. Introduced at 03028bb23ef29961f9934641df69c94609d2ed34. Also when testing beta versions, please note that betas use different configuration folder than full versions.

TurboCooler, 2020-12-13:
> Please try also new beta (3) when its released. There will be improvements in error reporting. Introduced at [03028bb](https://github.com/prusa3d/PrusaSlicer/commit/03028bb23ef29961f9934641df69c94609d2ed34). Also when testing beta versions, please note that betas use different configuration folder than full versions.

Just tried on 2.3 Beta 3 and it is working correctly and you get a good human readable error message as well.

bubnikv, 2020-12-13:
@TurboCooler What error message are you getting then?

TurboCooler, 2020-12-13:
> @TurboCooler What error message are you getting then?

The same one I posted above.  Previously, it was the error pictured in the original post.

rcmaniac25, 2021-01-08:
Same issue as OP. Just hit it today with the release of Prusament PVB.

bbolgradov, 2021-02-28:
I have similar issue and I've narrowed it down to incompatibility with ecryptfs.

Here is how I can reproduce it:

First test without encryption, works as expected:
```
!w / --------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> sudo mkdir /foo

!w / --------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> sudo chown boris:boris /foo

!w / --------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> prusa-slicer --loglevel 5 --datadir /foo/prusa-plain    
[2021-02-28 16:11:48.976695] [0x00007ff3f71d9080] [debug]   full path: 5969277306875100478
[2021-02-28 16:11:48.976707] [0x00007ff3f71d9080] [debug]   single instance: undefined. other params: prusa-slicer;--loglevel;5;--datadir;/foo/prusa-plain
[2021-02-28 16:11:48.976711] [0x00007ff3f71d9080] [debug]   full lock path: /foo/prusa-plain/cache/5969277306875100478.lock
[2021-02-28 16:11:48.976714] [0x00007ff3f71d9080] [debug]   get_lock(): datadir does not exist yet, creating...
[2021-02-28 16:11:48.976764] [0x00007ff3f71d9080] [info]    instance check: Another instance not found or single-instance not set.
[2021-02-28 16:11:49.200419] [0x00007ff3f71d9080] [info]    Checking if indices need to be installed from resources...
[2021-02-28 16:11:49.200460] [0x00007ff3f71d9080] [info]    Install index from resources: "FLSun.idx"
[2021-02-28 16:11:49.200471] [0x00007ff3f71d9080] [debug]   PresetUpdater: Copying "/usr/share/PrusaSlicer/profiles/FLSun.idx" -> "/foo/prusa-plain/cache/FLSun.idx"
[2021-02-28 16:11:49.209807] [0x00007ff3f71d9080] [info]    Install index from resources: "Creality.idx"
[2021-02-28 16:11:49.209815] [0x00007ff3f71d9080] [debug]   PresetUpdater: Copying "/usr/share/PrusaSlicer/profiles/Creality.idx" -> "/foo/prusa-plain/cache/Creality.idx"
[2021-02-28 16:11:49.212494] [0x00007ff3f71d9080] [info]    Install index from resources: "LulzBot.idx"
[2021-02-28 16:11:49.212500] [0x00007ff3f71d9080] [debug]   PresetUpdater: Copying "/usr/share/PrusaSlicer/profiles/LulzBot.idx" -> "/foo/prusa-plain/cache/LulzBot.idx"
[2021-02-28 16:11:49.215191] [0x00007ff3f71d9080] [info]    Install index from resources: "BIBO.idx"
[2021-02-28 16:11:49.215198] [0x00007ff3f71d9080] [debug]   PresetUpdater: Copying "/usr/share/PrusaSlicer/profiles/BIBO.idx" -> "/foo/prusa-plain/cache/BIBO.idx"
[2021-02-28 16:11:49.218033] [0x00007ff3f71d9080] [info]    Install index from resources: "TriLAB.idx"
[2021-02-28 16:11:49.218040] [0x00007ff3f71d9080] [debug]   PresetUpdater: Copying "/usr/share/PrusaSlicer/profiles/TriLAB.idx" -> "/foo/prusa-plain/cache/TriLAB.idx"
[2021-02-28 16:11:49.220703] [0x00007ff3f71d9080] [info]    Install index from resources: "Anycubic.idx"
[2021-02-28 16:11:49.220710] [0x00007ff3f71d9080] [debug]   PresetUpdater: Copying "/usr/share/PrusaSlicer/profiles/Anycubic.idx" -> "/foo/prusa-plain/cache/Anycubic.idx"
[2021-02-28 16:11:49.223502] [0x00007ff3f71d9080] [info]    Install index from resources: "PrusaResearch.idx"
[2021-02-28 16:11:49.223508] [0x00007ff3f71d9080] [debug]   PresetUpdater: Copying "/usr/share/PrusaSlicer/profiles/PrusaResearch.idx" -> "/foo/prusa-plain/cache/PrusaResearch.idx"
[2021-02-28 16:11:49.226599] [0x00007ff3f71d9080] [trace]   System language detected (user locales and such): en_US
[2021-02-28 16:11:49.229356] [0x00007ff3f71d9080] [trace]   Best translation language detected (may be different from user locales): en_US
[2021-02-28 16:11:49.229364] [0x00007ff3f71d9080] [trace]   Switching wxLocales to en_US
[2021-02-28 16:11:49.354514] [0x00007ff3f71d9080] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 111MB; Shared memory: 71MB; Private memory: 40MB; Peak memory usage: 111MB
[2021-02-28 16:11:49.355354] [0x00007ff3f0b38640] [trace]   Dbus object /com/prusa3d/prusaslicer/InstanceCheck/Object5969277306875100478 registered. Starting listening for messages.

(prusa-slicer:11084): Gtk-WARNING **: 16:11:49.449: Negative content height -13 (allocation 1, extents 7x7) while allocating gadget (node button, owner GtkToggleButton)

...snip...

(prusa-slicer:11084): Gtk-CRITICAL **: 16:11:49.452: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
04:11:49 PM: Debug: window wxTreeCtrl(0x55cce8e206c0, ) lost focus even though it didn't have it
04:11:49 PM: Debug: window wxTreeCtrl(0x55cce8e206c0, ) lost focus even though it didn't have it
04:11:49 PM: Debug: window wxTreeCtrl(0x55cce8e206c0, ) lost focus even though it didn't have it
04:11:49 PM: Debug: window wxTreeCtrl(0x55cce8e206c0, ) lost focus even though it didn't have it
04:11:49 PM: Debug: window wxTreeCtrl(0x55cce8e206c0, ) lost focus even though it didn't have it
04:11:49 PM: Debug: window wxTreeCtrl(0x55cce8e206c0, ) lost focus even though it didn't have it
04:11:49 PM: Debug: window wxTreeCtrl(0x55cce8e206c0, ) lost focus even though it didn't have it
04:11:49 PM: Debug: window wxTreeCtrl(0x55cce8e206c0, ) lost focus even though it didn't have it
04:11:49 PM: Debug: window wxTreeCtrl(0x55cce8e206c0, ) lost focus even though it didn't have it
[... cut ...]
[2021-02-28 16:12:32.891641] [0x00007f67da340080] [info]    Install index from resources: "FLSun.idx"
[2021-02-28 16:12:32.891654] [0x00007f67da340080] [debug]   PresetUpdater: Copying "/usr/share/PrusaSlicer/profiles/FLSun.idx" -> "/foo/prusa-encrypted/cache/FLSun.idx"
[2021-02-28 16:12:32.891789] [0x00007f67da340080] [error]   Uncaught exception: boost::filesystem::copy_file: Invalid argument: "/usr/share/PrusaSlicer/profiles/FLSun.idx", "/foo/prusa-encrypted/cache/FLSun.idx"
*** Caught unhandled unknown exception; terminating
```

Do you want me to create a new issue for this?

bubnikv, 2021-03-01:
> Do you want me to create a new issue for this?

I don't know. Frankly we cannot support everything. At least a fine grained analysis of what is going wrong, or better a pull request would be helpful.

bbolgradov, 2021-03-01:
If I have some time I might debug more. It worked with previous versions though, I don't know what changed in the slicer.

Hagb, 2021-05-20:
> I have similar issue and I've narrowed it down to incompatibility with ecryptfs.
> 
> Here is how I can reproduce it:
> 
> First test without encryption, works as expected:
> 
> ...

@bbolgradov 
https://github.com/boostorg/filesystem/commit/4b9052f1e0b2acf625e8247582f44acdcc78a4ce should fix it.

bubnikv, 2021-06-14:
We are now having another report of the same issue:
https://github.com/prusa3d/PrusaSlicer/issues/6588

> https://github.com/boostorg/filesystem/commit/4b9052f1e0b2acf625e8247582f44acdcc78a4ce

Yeah, this should fix it.

The boost::filesystem::copy_file is a mess of defines and various Linux APIs and workarounds for the bugs / incompatibilities. We will likely have to extract boost::filesystem::copy_file to our code base to have all the bug fixes even on older distros.

bubnikv, 2021-06-14:
@Hagb thanks for identifying the issue and fixing it in boost library.

At the end of the day, we will most likely maintain our own copy_file_linux() due to the issues of copy_file() sometimes failing at the very end setting target file attributes. I was not so lucky convincing boost guys that there is an issue to be fixed, thus we made  a copy of the function for our purposes.
2f8190e6d027fd748b17a34280b6811099f95e2d

We will most likely use our own copy_file implementation across the application.

closed 2021-06-28 by kocikdav

bubnikv, 2021-07-02:
Fixed with PrusaSlicer 2.3.2-beta by back-porting the boost::filesystem fix into PrusaSlicer own copy_file implementation.

commit 2f8190e6 (2021-06-14): Workaround of boost::filesystem::copy_file() incompatibility
on some file systems (eCrypt ...)
Should fix #4716 #6588
commit 1cb5b94c (2021-06-14): Workaround of boost::filesystem::copy_file() incompatibility
on some file systems (eCrypt ...)
Should fix #4716 #6588
commit f60f08fc (2021-06-17): Workaround of boost::filesystem::copy_file() incompatibility
on some file systems (eCrypt ...)
Should fix #4716 #6588
commit 85d4e1c6 (2021-06-28): Workaround of boost::filesystem::copy_file() incompatibility
on some file systems (eCrypt ...)
Should fix #4716 #6588