prusa3d/PrusaSlicer #4717: 2.3.0-Alpha Splash Screen

malcinator, 2020-09-08:
### Version
2.3.0 - Alpha

### Operating system type + version
macOS 10.15.6

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk2.5s * i3 Mk3S

### Behavior
**Splash Screen?** Why do we now after all this time need a Splash Screen? I can see from my Dock that the application is loading. I'm not sure what else I need to watch while its loading?

Will there be an option in General Preferences to disable it? If not can I request it?

**Yes** this is a request to turn off the new feature.

rtyr, 2020-09-08:
Unfortunately, starting on Windows OS can take a few seconds without any visible action, which can be confusing for some users.

Why is a splash screen issue for you?

Related issue https://github.com/prusa3d/PrusaSlicer/issues/4468

malcinator, 2020-09-08:
Splash-screens are few and very far between on macOS. Opening a Splash-screen to tell me an application is loading is really unnecessary as the feedback from the Dock icon is plenty to inform me that the application is indeed opening. If such a splash-screen is required on Windows the please only show it on windows.

lukasmatena, 2020-09-11:
The splash screen can be turned off in Preferences since f1f9785a8a3d10ad7a3bb91194e0d758cab5aee3.
Closing.

closed 2020-09-11 by lukasmatena