prusa3d/PrusaSlicer #4719: Alpha 2.3.0 Preview not updating

Area5142, 2020-09-09:
### Version
PrusaSlicer Version:   version_2.3.0-alpha0-789-ge010d287c (and earlier alpha versions)
Build:     From source

### Operating system type + version
Linux Mint 19.3 x64

### Behavior
Preview is not updating and blank build plate is shown.

Configuration: Background processing is activated.

Steps to reproduce error:
1) Start PS.
2) Right-click build plate and add Shape->Box.
3) Press CTRL-6.
4) Empty build plate is shown.

Pressing (Re)Slice Now has no effect.

(De)Selecting Perimeter in the new "Estimated printing time..." dialog forces an update but the output is distorted as shown here:

[image]

fiferboy, 2020-09-10:
Same thing is happening here on a fresh build on Ubuntu 20.04 - but toggling "Brim" in the print settings pane seems to fix it for the rest of the session. In fact, changing pretty much any print parameter seems to fix the problem, but it is very annoying when it happens in the first place.

Area5142, 2020-10-14:
Just tried latest build: version_2.3.0-alpha0-1138-g6eee0c043

Build plate is still blank when switching to preview and Background processing is on.
Need to switch Brim on and off to force an update as @fiferboy suggests.
OK it works but is annoying...

foreachthing, 2020-10-15:
Same here on Win10x64.

bubnikv, 2020-10-15:
@enricoturri1966 will look into this. Thanks for testing.

Area5142, 2020-10-15:
On Mint Linux 19.3 x64 the redraw problem is now fixed in PS version_2.3.0-alpha0-1150-g58d57f9b7 and toolpaths is shown without distortion. Thank you.

If @foreachthing can say the same for Win 10x64 the issue can be closed.

foreachthing, 2020-10-15:
@Area5142 , OK! It seems to work!
Didn't extensively test it, but what didn't work before, works now.

closed 2020-10-16 by enricoturri1966