prusa3d/PrusaSlicer #4720: Alpha 2.3.0 compile error: Symlinking the G-code viewer to PrusaSlicer

Area5142, 2020-09-09:
### Version
version_2.3.0-alpha0-789-ge010d287c

Compiled from source.

### Operating system type + version
Linux Mint 19.3 x64

### Behavior
Make fails with the error:
```
Symlinking the G-code viewer to PrusaSlicer
/bin/sh: 1: cd: can't cd to /home/<user home removed>/src/PrusaSlicer/Build/src/$<TARGET_FILE_DIR:PrusaSlicer>
src/CMakeFiles/PrusaSlicer.dir/build.make:146: recipe for target 'src/prusa-slicer' failed
make[2]: *** [src/prusa-slicer] Error 2
make[2]: *** Deleting file 'src/prusa-slicer'
CMakeFiles/Makefile2:216: recipe for target 'src/CMakeFiles/PrusaSlicer.dir/all' failed
make[1]: *** [src/CMakeFiles/PrusaSlicer.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

I am just trying to compile the binary for encapsulation into AppImage container.

Editing the src/CMakeList.txt file and commenting out:

```
if (APPLE)
        # On OSX, the name of the binary matches the name of the Application.
        add_custom_command(TARGET PrusaSlicer POST_BUILD
            COMMAND ln -sf PrusaSlicer prusa-slicer
            COMMAND ln -sf PrusaSlicer prusa-gcodeviewer
            COMMAND ln -sf PrusaSlicer PrusaGCodeViewer
            WORKING_DIRECTORY "$<TARGET_FILE_DIR:PrusaSlicer>"
            COMMENT "Symlinking the G-code viewer to PrusaSlicer, symlinking to prusa-slicer and prusa-gcodeviewer"
            VERBATIM)
    else ()
#        add_custom_command(TARGET PrusaSlicer POST_BUILD
#            COMMAND ln -sf prusa-slicer prusa-gcodeviewer
#            WORKING_DIRECTORY "$<TARGET_FILE_DIR:PrusaSlicer>"
#            COMMENT "Symlinking the G-code viewer to PrusaSlicer"
#            VERBATIM)
    endif ()

```
Makes me compile the source. What is the purpose of this symlink?

_Is this a new feature request?_ No

bubnikv, 2020-10-05:
symlink is there so that one can start PrusaSlicer in a G-code viewer mode without adding a --gcodeviewer parameter.
It is simulating one application with the other by giving it a different name through the symlink.

lukasmatena, 2020-10-05:
As to the error, we use a cmake generator expression when making the symlink, which is apparently only supported since cmake 3.13 in this context. Instead of doing it differently, we will use this as a reason to update minimal required cmake version to 3.13 (already done by 4d3190f). Symlink or not, we would have done this anyway at one point. The build process is already complicated enough, it makes sense to at least assume something about cmake. 3.13 is out for almost two years, so it makes sense to move forward.

Closing the issue.

closed 2020-10-05 by lukasmatena

Area5142, 2020-10-06:
On Linux Mint 19.3 the default repository cmake version is 3.10.2 and that was the problem.
To install newest cmake from apt.kitware.com run the following:

```
wget -O - https://apt.kitware.com/keys/kitware-archive-latest.asc 2>/dev/null | gpg --dearmor - | sudo tee /etc/apt/trusted.gpg.d/kitware.gpg >/dev/null
sudo apt-add-repository 'deb https://apt.kitware.com/ubuntu/ bionic main'

```
This will install the public key from kitware and add reference to the repository.

```
sudo apt update
sudo apt upgrade

```
This will upgrade the installed cmake to version 3.18.3.

Compiling latest 'version_2.3.0-alpha0-1019-gf47ad1fd9' succeeds and a link 'prusa-gcodeviewer' is created to 'prusa-slicer.'

I am still in the dark on, how to pack this in the AppImage, supporting both prusa-slicer and prusa-gcodeviewer. Maybe a link with the right name to the AppImage will solve this...

bubnikv, 2020-10-07:
On AppImage, you have to start each application instance by executing the
appimage, see the discussion starting with
https://github.com/AppImage/AppImageKit/issues/419#issuecomment-685786115

Our AppImage will support symlinking the whole appimage from the outside of
the appimage as prusa-gcodeviewer.

Area5142, 2020-10-08:
Thank you for explaining how to make gcodeviewer work for the AppImage.

Latest alpha version_2.3.0-alpha0-1055-g6871c2ccd seems not to support the symlink to prusa-gcodeviewer yet.