prusa3d/PrusaSlicer #4721: [Feature Request] Indicate Sequential Printing better in preview screen

Qqwy, 2020-09-09:
### Version
PrusaSlicer 2.2.0

### Operating system type + version
Manjaro Linux x86-64

### Behavior

PrusaSlicer has a great capability to print models sequentially.
However, when previewing what the printer will do in the 'preview' screen, it is not clear how (or even if) the models are printed one after the other:

- There is a time estimation only for the whole print.
  - We could add times per model here.
- The vertical 'layer slider' works on layers for all models together.
  - We could 'split' the layer slider into multiple parts bottom-to-top per model. Then the slider as a whole would function as an indication of 'time' (measured in total layers), whereas each of the sub-sliders would function per-model.

neophyl, 2020-09-09:
So how would that sub slider work when you have 96 sequential print objects ?  That’s not an idle example as I have printed several plates with 96 items on. 

You can see the print order by using E to show the numbering but I agree that a proper preview mode where you can see what the extruder is doing on each layer would be much better. The old Prusa Control software used to do that.

Area5142, 2020-09-09:
In the alpha version there is an option with a slider to see the tool movement on each layer.

bubnikv, 2020-11-21:
PrusaSlicer 2.3.0-alpha4 implements visualization of sequential print by the G-code viewer.

closed 2020-11-21 by bubnikv