prusa3d/PrusaSlicer #4722: bug: support stuctures is generated on neighboring objects

mgx0, 2020-09-10:
### Version
2.2.0+

### Operating system type + version
MacOS 10.14.6

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S

### Behavior
- assume you have two object near themselves (print in place flexible thing)
- add support enforcer to one object
- PS adds single-layer support structure to the printing plate under second object. not around it - UNDER it

#### Project File (.3MF) where problem occurs

[image]
[image]
[image]

[testprint.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5200132/testprint.3mf.zip)

foreachthing, 2020-09-10:
If I encounter this issue, I model the required support(s) in CAD.
Or you could use a support blocker (modifier).

mgx0, 2020-09-10:
> If I encounter this issue, I model the required support(s) in CAD.
> Or you could use a support blocker (modifier).

Yeah but I still think it's a bug and should be solved.

mgx0, 2020-09-10:
[testprint2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5200666/testprint2.3mf.zip)

reopening as correcting the parts did not help.

this parts gets automatic support
[image]

this part does not
[image]

but first part's supports are laid under second part
[image]

and support blocker does not help 
[image]
[image]

neophyl, 2020-09-10:
Isn't this because each object is treated as a completely independent entity so supports for one object take no notice of any other object ?  It's my understanding that the slicer is designed this way on purpose to allow all the things like variable layer heights for different objects etc.

In this case I think the correct procedure is to place the first object on the plater and then right click it and 'add part' to add the second part.  Then place them where you need them.  This makes both models part of the same logical object.  Any support then generated will take both models into consideration.

mgx0, 2020-09-10:
that one makes sense and works. is this issue to be closed now? as it seems it's not a bug but my lack of knowledge

n8bot, 2020-09-10:
This is definitely a bug. Sure, you can work around it in silly ways. But, this is a bug.

mgx0, 2020-09-10:
well support blocker does not work either. And it does not matter which object I add it to, it simply creates the fist layer of supports anyway. I'll leave it open and let responsible folks deal with it. Now I know how to work around it, mission accomplished, thanks @neophyl

foreachthing, 2020-09-11:
Question: are these separat parts (multiple STL, 3mf?) you assemble in PS?
Maybe make one single object (CAD, Blender, ...) from your parts and load that into PS.

mgx0, 2020-09-11:
Yeah it's two stl files. I know having it as a single stl helps but this is still a bug to me

bubnikv, 2021-02-24:
Support structures are generated independently for each object by PrusaSlicer. That is different from all other slicers.

We should handle the overlapping rafts, there is a duplicate issue for that https://github.com/prusa3d/PrusaSlicer/issues/4517

For your particular case, you should merge all the parts of a single object that belong together into a single object. Select them in Plater, right mouse click opens a pop-up menu. Select merge.

closed 2021-02-24 by bubnikv