prusa3d/PrusaSlicer #4726: Print Host Upload - Unhandled Exception

blue-hound, 2020-09-10:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 Version 1909 (OS Build 18363.1016)

### 3D printer brand / version + firmware version (if known)
MK3S MMU2S+ 3.9.0 (1.0.6)

### Behavior
When I setup the FlashAir in the print host upload screen, it uploaded the files to the root directory. I have mapped a windows share to a directory on the FlashAir called G-Code. I wanted to get PrusaSlicer to upload to this directory so I added /G-Code to the end of the host. When I clicked on the test button it caused an unhandled exception.

```
11:37:39: Internal error: boost::too_many_args: format-string referred to fewer arguments than were passed
11:37:39: Internal error: boost::too_many_args: format-string referred to fewer arguments than were passed
```

#### Project File (.3MF) where problem occurs
Not applicable.

[image]
[image]

maxcom99, 2020-10-06:
Also getting this error  

[image]

blue-hound, 2020-11-02:
Just wanted to add to this, if the IP address doesn't connect (e.g. the printer is turned off), this error also occurs.

lukasmatena, 2020-11-02:
@maxcom99 @blue-hound Could you guys please retest with PrusaSlicer 2.3.0-alpha2? I cannot reproduce it, only in 2.2.0. I think this may have been already fixed by @YuSanka when she implemented the concept of physical printers.

blue-hound, 2020-11-25:
Will do. Been away for a bit, I'll check out the latest development version in the next day or so and get back to you.

blue-hound, 2020-11-30:
Hi just tested this with PrusaSlicer-2.3.0-beta1+win64-202011281033 and the error appears to be handled properly. I couldn't reproduce this.

Thanks for looking into this!

closed 2020-11-30 by blue-hound

lukasmatena, 2020-12-01:
Thanks for retesting this.