prusa3d/PrusaSlicer #4727: po instalaci na windows 10 aplikace prusaslicer padá

Shootercz, 2020-09-10:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-09-10:
Můžete prosím napsat více detailů? Jak se to konkrétně projevuje, objevuje se nějaká chybová hláška? Nějaké základní informace o PC, verze PrusaSliceru apod.

Shootercz, 2020-09-12:
Žádná chybová hláška se neobjeví, aplikace se po 2s až 2min zavře. Aplikaci jsem zkoušel reinstalovat a smazat konfigurační soubor.

[image]
[Prusaslicer.txt](https://github.com/prusa3d/PrusaSlicer/files/5212892/Prusaslicer.txt)

Shootercz, 2020-09-12:
Žádná chybová hláška se neobjeví, aplikace se po 2s až 2min zavře. Aplikaci
jsem zkoušel reinstalovat a smazat konfigurační soubor.

PrusaSlicer
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 8,433MB
OpenGL installation
GL version:   4.6.0 NVIDIA 451.67
Vendor:       NVIDIA Corporation
Renderer:     GeForce MX150/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
[... 312 lines cut ...]
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

lukasmatena, 2020-09-17:
@Shootercz Zdravím. Nejste první, kdo hlásí sporadické crashe na Win10. Pokud vím, problém se nám nepodařilo reprodukovat a nevíme, čím je to způsobeno. Je ve Vašich silách zkompilovat PrusaSlicer v debug módu a dostat z toho nějaký backtrace?

Pokud ne, možná bychom Vám mohli poslat PDB build. @bubnikv ?

Shootercz, 2020-10-08:
Dobrý den, problém přetrvává, posílám výpis debugu z console. 
[Prusaslicer2.txt](https://github.com/prusa3d/PrusaSlicer/files/5350771/Prusaslicer2.txt)

rtyr, 2021-02-24:
Stává se to i s aktuální verzí 2.3.0 a na aktualizovaném Win10?

bubnikv, 2021-09-25:
bez odpovědi, zavíráme.

closed 2021-09-25 by bubnikv