prusa3d/PrusaSlicer #4729: [Feature Request] Retract Options for Sequential Printing

SphaeroX, 2020-09-11:
### Version
Version 2.2.0+win64

### Operating system type + version
Microsoft Windows 10 Pro [Version 10.0.18363.959]

### 3D printer brand / version + firmware version (if known)
Custom Klipper

### Behavior
With Sequential Printing you can choose all kinds of options, with the Height Range Modifier you can make a perfect Heattower. If you could now control the retract in Sequential Printing, the slicer would be the ultimate tool to calibrate your printer and find out the best settings.

rtyr, 2020-09-11:
It looks like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3635.

SphaeroX, 2020-09-11:
> It looks like a duplicate of #3635.

You are right! Sorry.

closed 2020-09-11 by SphaeroX