prusa3d/PrusaSlicer #4730: Improved zooming in PrusaSlicer

DanskDave, 2020-09-11:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 Pro version 2004 (Build 19041.508)

### 3D printer brand / version + firmware version (if known)
Original Prusa MINI

### This is a new request feature
Hello
I have a suggestion to improve zooming in PrusaSlicer!
I'm suggesting to zoom in/out where the 'cursor' is and not based on the 'Spring steel sheet' so you can zoom in/out where you want to.
This zooming option is standard in SketchUp Pro which I use a lot.
See file attached (SketchUp Pro 2018.mp4)
This is in comparison to what PrusaSlicer does now.
See file attached (PrusaSlicer.mp4)
I have reduced the resolution of both files to make the file size smaller.
I hope this makes sense and I love using PrusaSlicer so keep up the good work!
Best wishes
Dave

[PrusaSlicer - Improved zooming in PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/5207199/PrusaSlicer.-.Improved.zooming.in.PrusaSlicer.zip)

foreachthing, 2020-09-11:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4417 ?

neophyl, 2020-09-11:
Yes its a duplicate.  
I have answered a few threads on the Prusa forum related to this, mainly where users don't realise you can pan the centre of the view around using the right mouse button and then zoom in to the area in the middle of the view.  They complain they cant zoom into something on the edge of the plater when in reality you can once you know how.

I actually find the PS way (once you get used to it) easier to use than the zoom to mouse cursor, especially when you have the mouse sensitivity very high like I usually do.

foreachthing, 2020-09-11:
I like zoom to cursor much better :wink:
Like in SolidWorks, Blender ...

Actually, as long as I can use my 3DConnexion device i'm good :grin:

neophyl, 2020-09-11:
Ah but Blender its configurable :) (like almost everything in that software) and I use zoom to selected object.  

My point is that changing existing operation without making it a configurable option might make it better for some people but for those who are happy with how it works now, well it will mess them up.

A lot of people don't realise that its possible to pan at all in PS in the first place.

foreachthing, 2020-09-11:
> Ah but Blender its configurable :) (like almost everything in that software) and I use zoom to selected object.

You're right! I forgot, because I transfer my settings from release to release.

> My point is that changing existing operation without making it a configurable option might make it better for some people but for those who are happy with how it works now, well it will mess them up.

Well, yeah. Don't change a current behavior - make it configurable. But, that's just my personal opinion!
I'm used to zoom to cursor and if one software does not allow that ... it vexes me.

> A lot of people don't realise that its possible to pan at all in PS in the first place.

Hard to imagine! Really?!? :worried:
Huh! It's even documented, I did not expect that! https://help.prusa3d.com/en/article/view_1761
But only on the website, not in shortcuts `?` dialogbox.

n8bot, 2020-09-11:
I would also welcome the change, as suggested, of a "cursor target" to pivot/zoom into. I prefer Fusion 360's way: A green dot is shown on the model/scene where the pivot is. To change the pivot, middle-mouse-click somewhere on the scene and the pivot will be set there.

Another suggestion is to change the FOV for perspective camera to something like 50 degrees ( I suspect it is ~90 right now), as well as allow the user to zoom much closer into the object than is currently allowed.

mark-hahn, 2021-07-26:
It is painful to use the zoom to cursor on fusion and most other cad programs and then bring up pslicer and not have it.  Even in most 2d zooming on the web it is to the cursor.  This defect is the one thing I bitch about every time I use pslicer.

There are logical arguments also.  With zoom to cursor it is easy to use the center of the bad and make it behave like it does now.  But with the current type of zooming it is impossible to zoom to cursor.

The second logical argument is simple.  It takes two operations to pan and zoom and only one operation to zoom to cursor.  Also, I find I have to adjust the pan over and over as I zoom.

mikolaszuza, 2021-07-28:
I know it's not an answer to your feature request, but try using the **Z key** ;) It zooms to selected object/s or to all objects if none is selected.
And you can change the center of Zoom with the right mouse button, as mentioned above.

mark-hahn, 2021-07-28:
Thanks.  That looks useful.  I'll have to look up all the shortcuts.  But
yes, that doesn't help the ease-of-use of zooming into a feature close-up.

altuor, 2021-12-25:
Definitely would be usefull to zoom into mouse cursor as an option (my initial request #4417), workarounds are nice and I learned a bit to use them, still though ;-)

Lajbror, 2022-08-08:
Please, make it possible to zoom into mouse cursor. Almost all modeling programs I know are able to do that. Preserve the old method for those users that are happy with it and for the love of God make it possible to zoom where the mouse cursor points :-) I love PrusaSlicer and this is one of those things that are necessary for power users and would make this slicer a lot better.

mark-hahn, 2022-08-08:
> Almost all modeling programs I know are able to do that

And almost all 2D zooming like google maps on the web.  For God's sake fix this.

stepikovo, 2022-08-22:
I'm closing this as a duplicate of #4417, let's keep GitHub clean :)

closed 2022-08-22 by stepikovo