prusa3d/PrusaSlicer #4732: Crash with "Spiral vase" mode

verglor, 2020-09-11:
### Version
_2.2.0_

### Operating system type + version
_Windows 10 b19042_

### 3D printer brand / version + firmware version (if known)
_Stock Ender 3 @ Marlin 2.6.0.1_

### Behavior
* Application crash after clicking _"Slice now"_ with _"Spiral vase"_ mode enabled on attached model.

#### Project File (.3MF) where problem occurs

[Fast Bed Level Calibration.zip](https://github.com/prusa3d/PrusaSlicer/files/5209050/Fast.Bed.Level.Calibration.zip)

rtyr, 2020-09-11:
Reproduced, altough the attached model is only 0.2mm high (1 layer print).

hejllukas, 2020-12-09:
It should be fixed with https://github.com/prusa3d/PrusaSlicer/commit/0ad135ff9314ba92a329d3f0df6117ecb71a58c5.

closed 2020-12-09 by hejllukas