prusa3d/PrusaSlicer #4736: Brim default behavior is suboptimal

Uko, 2020-09-12:
This is sort of a feature request that comes from the observation of comments on Reddit & Facebook as well as my personal experience.

Over time PrusaSlicer got many new features, and some of them interfere with ease of use. Because of the apparent complications, people are asking questions and reporting bugs. In turn, developers reply that this is a feature and not a bug, completely dismissing the UX part of the issue.

In particular, I'm talking about **brim**. This is such a basic feature that the brim checkbox is available on the "plater" sections. Whenever someone enables brim, they will notice two issues (and often write about them):
  - the skirt is printed inside of the brim
  - there is a gap between the printed object and the brim

Personally, each time when I tick the brim checkbox, I also change the distance between the object and the skirt, so it is on the outer edge of the brim, and I set the elephant foot compensation to 0.

One suggestion that I have is to change these values automatically when the brim checkbox is ticked. In my opinion, at least this should be done for the elephant foot compensation because otherwise, you select a brim, so your print sticks to the build plate, and the brim does not even hold the object.

Although there may be many ways to solve the described problem, I want to highlight, that the purpose of this issue report is not to say that some issue is wrong but to point out that the basic user experience is suboptimal

louwheesy, 2020-09-12:
Personally, I have noticed that even with elephants foot compensation that the brim will adhere very well to the part.

I definitely do agree that the default skirt behavior needs to be worked on.

lukasmatena, 2020-09-12:
> PrusaSlicer got many new features, and some of them interfere with ease of use

The new features are not added because the developers have nothing else to do, but they often rise from users' requests. As for the skirt-brim overlap, in older versions the skirt ignored its 'distance from object' settings when brim was used (it was pushed farther from the object). This led to two issues that users also "often wrote about".

1. It was impossible to use tall skirt as a draft shield.
2. Tall skirt (draft / ooze shields)would break off the bed, it had inadequate surface contact.

You can check out issue #724 as an example. And it was reported more times. That is why current behaviour was introduced - the skirt can be placed close to the object and the brim (if enabled) will stabilize it. Both issues are addressed by this. Your solution would reintroduce them.

As for the elephant foot, its purpose is to compensate for first layer squish, meaning the object is elsewhere when you see it in preview. That is why there is a gap in preview. If you set the elephant compensation value to match the actual elephant foot, it should work. And the tests than we have done prove it. You can find details in https://github.com/prusa3d/PrusaSlicer/issues/3949 and several linked issues. My personal impression is that people set elephant foot compensation / live z too high, which than leads to these issues. That is why the value is configurable.
You can see that @louwheesy just confirmed that this is fine from his point of view, and there is also an inverse issue where someone complains that the brim is too CLOSE to the object (https://github.com/prusa3d/PrusaSlicer/issues/3949).

I will not stubbornly insist that current behaviour is the best possible. I'm just saying that there is usually a reason for changes that are made and proposed improvements should take it into account.

> each time when I tick the brim checkbox, I also change the distance between the object and the skirt, so it is on the outer edge of the brim

Just out of curiosity. What is the purpose of skirt in this case?

Uko, 2020-09-12:
> Personally, I have noticed that even with elephants foot compensation that the brim will adhere very well to the part.
> 
> I definitely do agree that the default skirt behavior needs to be worked on.

It's true, there were many times when the brim worked for me without changing the elephants foot compensation, but there were also times when the brim just won't stick to the object. And actually, this photo on Facebook where someone is asking why the brim is not connecting to the parts that they print. (and before that there were many many similar questions)
[image]

louwheesy, 2020-09-12:
I have to echo what Luca said. Your elephant's foot compensation is set too high. Its not supposed to act like a chamfer, it's supposed to allow your printed dimensions to match your model's. So if your part is supposed to be 5mm wide and it comes out to be 5.2mm wide on the first layer, then you use elephant's foot compensation to bring the part back to 5mm. Your brim is going to start after those 5mm regardless. 

My problem with using a tall skirt as a draft shield is that it's not really the original intent of a skirt. I feel that the draft shield and skirt should be different tools because of the aforementioned possibility where it causes interruptions in the concentricity of the brim.

Uko, 2020-09-12:
@lukasmatena again, I'm not saying that the features are bad and should be reverted, I'm just asking whether the defaults are correct (after all when you enable supports generation the default behavior asks you if you want to detect bridging perimeters, although you could turn this setting on yourself, and although there were times when I disabled it).

Now let's focus on the defaults. I'd like to ask two questions:

### why do you want the skirt to be printed in the middle of a brim when it is enabled?
You would say that this is for tall skirts used as draft shields. But I don't see a draft shield when I enable brim. Maybe I'm wrong (and in fact, the Prusa company is the only entity with the authority to survey people about what they do the most), but I see questions about skirt being in the middle of a print much more often than people showing off how they used a draft shield. And also form my experience, there were only a few cases when I used a draft shield. This does not make the feature bad, I love it, and it helps a lot with draft shields. But having a skirt in the middle of the brim by default?

Why do I set it on the edge of a brim? Well, I don't like it in the middle, and one of the reasons is described in the last issue that you liked: https://github.com/prusa3d/PrusaSlicer/issues/3949#issuecomment-605613952 the brim is not printed well when there is a skirt inside of it. It's true, one can also disable the skirt altogether, this is also a good idea for default behavior. However, when I have a skirt on the outside it is easier to remove the brim from the build plate. For example when I print with TPU in a flat sheet it both manages to warp, and stick so well that it's hard to remove. But maybe this is a niche use case, maybe the default should be just to disable skirt.

 
### why do you want the brim to have a gap with the printed object?
You say that some people set the elephant foot compensation too high. Well, I don't set it. It is set by default. And yes, I can tweak everything, but imagine that someone buys a 3d printer, and tries to print something. It warps. Then they turn on the brim. It warps again because the brim does not attach well. These questions were not present when there was no _default_ elephant foot compensation in PrusaSlicer. Now when you have a default you can't blame people for setting it too high because they don't set it.

But even if we step away from the defaults. The description of the elephant foot compensation says:

> the first layer will be shrunk in the XY plane…

It should be changed to

> the first layer **except brim** will be shrunk in the XY plane…

And again, you don't make a gap between the perimeters and the infill for the elephant foot compensation, although the lines are not going to be in that place because you have a squish

Uko, 2020-09-12:
> I have to echo what Luca said. Your elephant's foot compensation is set too high. Its not supposed to act like a chamfer, it's supposed to allow your printed dimensions to match your model's. So if [...]

Well, in this case I have to accuse the PrusaSlices of setting the elephant foot compensation by default to 0.2, because usually I don't have an elephant's foot at all.

Look, I think that this default setting is there because many-many people who use PrusaSlicer over squish their first layer. And on average 0.2 mm of compensation is going to improve the prints more than ruin them. At the same time, many people who use brim with PrusaSlicer will have initial Z too high which in combination with the default elephant's foot compensation will result in the brim not attaching.

ingoclaro, 2020-09-14:
Hey, that is my picture :)
For me at least since I was using the defaults, it caught me by surprise that when activating brim, it was not attaching to the object and had no idea why, hence my question in facebook. Even a note on the elephant foot setting wouldn't solve this, since I activated brim, not elephant foot.

I have some suggestions to improve this:
1/ improve the (hover) description of brim so that it mentions that elephant foot compensation can mess with it
1b/ add a text below brim checkbox if there is elephant foot compensation >0, similar to wall width text when changing horizontal shells.
2/ perhaps add a "separation from object setting to the brim group (similar to the skirt group) that takes the elephant foot compensation into account, so if I put 0, it's indeed 0 after elephant foot compensation is applied.
3/ add a warning modal similar to the one that shows when enabling supports and user can choose if she wants to set elephant foot compensation to 0 or not.

lukasmatena, 2020-09-14:
@Uko 
Thanks for the clarification, I admit there are some good points.  I personally (based on my experience with one printer and one filament type) tend to agree that setting brim to e.g. 4mm and skirt to 5mm would really prevent this issue from happening for most users. And disabling skirt completely in the default profile would not hurt me either - I honestly don't really know what it supposedly does that the priming line can't. That is me, though. The profiles are managed by @rtyr, who knows details about why they look the way they do and is aware of all corner cases and so on.

As for the EFC, I don't know. As far as I know, it was off by default before it was improved for 2.2.0 release. You may be right that users who don't know what EFC is and do not set the live z properly might be caught by surprise if the default overcompensates. And it may even be aggravated by the fact that EFC settings are hidden in simple mode, so anyone using simple mode not only didn't set it wrong, but also cannot fix it. Maybe having it off by default (or set a more conservative value) so users have to break it themselves first would be better.

As for that @ingoclaro 's picture that you posted, I'm not sure what I should see, but it looks underextruded to me - so there is no surprise that the EF is overcompensated with the default 0.2mm correction.

@louwheesy 
>  I feel that the draft shield and skirt should be different tools because of the aforementioned possibility where it causes interruptions in the concentricity of the brim.

Even if they were a different tool in the UI, the problem of what to do when brim is also used does not solve itself. It will still intersect.

WhiteDragonD, 2020-09-23:
This was annoying because I sliced up a bunch of parts to print, started a few 14 hour prints only to notice the brim wasn't connected to the part.  Google searching brought me here.

The ONLY thing I do is change the skirt distance to 7 mm with the Brim at 5, so I am not happy that I just click the Brim box, slice, save.... and it doesn't actually attach.    That's a bug.

rtyr, 2020-09-29:
**Elephant foot compensation**
The default value is based on our tests and feedback. The value should be also as universal as possible with different sheet types (smooth/textured). Some people have actually the opposite opinion, that the default EFC value should be higher (usually TXT sheet users). I agree that elephant foot compensation setting should be more "visible" to the users.

**EFC vs brim**
Personally, I didn't have any issues with it yet. In my experience, the gap between the object and the brim is usually a result of incorrectly set first layer (live Z). The FB photo mentioned here is a good example. On the other hand, I can understand, that "the correct" first layer can differ slightly just because of personal preferences. Is the current behavior correct? I am not sure. I can imagine some issues on printers with uneven first layer (for whatever reason). I asked some of our tech support guys about brim/EFC feedback and it seems, that our customers doesn't have issues with the current behavior. There is usually only some confusion coming from the preview, where the gap is clearly visible.

_Elephant foot compensation 0/0.1/0.15/0.2/0.3:_
[image]

**Skirt vs brim**
I wouldn't mind disabling the skirt, but I can imagine it would be not so popular decision. Many people are using it for liveZ finetuning, for example. It is also better to stabilize nozzle pressure close to the main object to avoid some missing material at the beginning of the first perimeter.
As for the default skirt distance from the object. Is 4mm for the brim mostly used value? Or is it 5mm or 6mm, maybe 10mm for someone? It seems quite unlikely, that we can find one perfect number for default distance. Increasing the default distance would limit some users, which often print full plates.

Maybe something like this https://github.com/prusa3d/PrusaSlicer/issues/4802 would help.

jurassic73, 2020-10-13:
Thanks for raising this concern!  My search happily found this issue you raised.  If I need a brim to help my print adhere to the printing surface, elephant foot is not of concern on the first layer... I want that brim adhered to the first layer solidly.

lukasmatena, 2021-08-12:
Two issues have been raised here, both should be gone in PrusaSlicer 2.4.0-alpha1:
- EFC and brim: https://github.com/prusa3d/PrusaSlicer/issues/3870#issuecomment-897501111

closed 2021-08-12 by lukasmatena