prusa3d/PrusaSlicer #4739: richiesta di modifica di Prusa Slicer 2.2.0

carlomontecchi, 2020-09-12:
Serramazzoni Modena Italia 12/9/2020

Vi propongo una modifica ai tempi di cambio tool nel programma PrusaSlicer 2.2.0.
Ho inserito l'istruzione di cambio tool nel Gcode cambio attrezzo e mi da questo risultato.

G1 X 250.08 Y 461.16 E 158.20 ;ultima riga con estrusore precedente
G1 E 148.20 F2400 
G92 E 0.00 
G4 P200 
M150 R255 ;sollevamento estrusore
M380 
G4 P100 
M381 
M150 R0 
T0 ;cambio estrusore
M380 ; eccita Sol ;abbassamento nuovo estrusore
G4 P100 ; Pausa 
M381 ; Spegne Sol 
G1 X 82.04 Y 86.24 F3600 

Pero' il nuovo estrusore viene abbassato prima dello spostamento veloce alla nuova posizione di inizio col nuovo estrusore e gratta fino a perdere dei passi.
Pertanto la sequenza dovrebbe diventare cosi':

G1 X 250.08 Y 461.16 E 158.20 ;ultima riga con estrusore precedente
G1 E 148.20 F2400 
G92 E 0.00 
G4 P200 
M150 R255 ;sollevamento estrusore
T0 ;cambio estrusore
M380 ; eccita Sol ;abbassamento nuovo estrusore
G4 P100 ; Pausa 
M381 ; Spegne Sol 
G1 X 82.04 Y 86.24 F3600 
M380 
G4 P100 
M381 
M150 R0 

Vorrei che vi inventaste l'opzione di diseccitazione solenoide dopo ultima riga del vecchio estrusore ed eccitazione del solenoide dopo la prima riga del nuovo estrusore.
Grazie per la comprensione.
Carlo Montecchi

wavexx, 2020-09-12:
English would be helpful, if possible?

carlomontecchi, 2020-09-13:
Serramazzoni Modena Italy 12/9/2020

I propose a change to the tool change times in the PrusaSlicer 2.2.0
program.
I inserted the tool change instruction in the tool change Gcode and it
gives me this result.

G1 X 250.08 Y 461.16 E 158.20; last row with previous extruder
G1 E 148.20 F2400
G92 AND 0.00
G4 P200
M150 R255; extruder lifting
M380
G4 P100
M381
M150 R0
T0; extruder change
M380; excites Sol; new extruder lowering
G4 P100; Pause
M381; Turns off Sol
G1 X 82.04 Y 86.24 F3600

However, the new extruder is lowered before moving quickly to the new
starting position with the new extruder and scratches until steps are lost.
Therefore the sequence should become like this:

G1 X 250.08 Y 461.16 E 158.20; last row with previous extruder
G1 E 148.20 F2400
G92 AND 0.00
G4 P200
M150 R255; extruder lifting
T0; extruder change
M380; excites Sol; new extruder lowering
G4 P100; Pause
M381; Turns off Sol
G1 X 82.04 Y 86.24 F3600
M380
G4 P100
M381
M150 R0

I would like you to invent the option to de-energize solenoid after last
row of the old extruder and energize solenoid after the first row of the
new extruder.
Thanks for understanding.
Carlo Montecchi

closed 2026-09-29 by prusaslicer-bot[bot]