prusa3d/PrusaSlicer #4744: Add a modifier to block brims / Add possibility to add custom brims

nils-kal, 2020-09-12:
### Version
2.2.0

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
any

### Behavior
I have a few models where I would need not to have a brim on certain spots (like when there is another layer directly going over the brim and its not possible to remove the brim afterwards).
Also the brim will sometimes be placed inside the model, like when the scructure is not fully closed in all places. 
Sometimes you just don‘t want to have brims on certain spots because it will be fiddly to remove it on this spot.

It would be great to either have the possibilty of blocking a brim with a modifier (support blockers do not block brims) or also the possibility to place custom brims with a modifier.

[image]

wavexx, 2020-09-13:
Custom brims using modifiers make a *lot* of sense. And would be easy to use to create "ears" where needed (just add a cylinder!). 

I'm currently adding parts to improve adhesion where needed, but since this causes the first layer to be sliced along with the extra parts, the perimeters will "bleed" into them, making them harder to remove.

nils-kal, 2020-09-13:
I know I can get mouse ears by adding a cylinder but I don‘t like this solution as this will also change where the perimeters are running (its rather changing the geometry of the part than having a custom brim somewhere). After removing them it gives a oddish look to the part

Iqwertz, 2020-11-18:
Yeah! would be really great if it could be implemented!

GeekDad63, 2020-12-18:
I often have several parts on the bed but only one or two need a brim. It would be nice to be able to add a brim to only the objects that need them.

fe60, 2020-12-23:
**Brim blockers** have been requested already in https://github.com/prusa3d/PrusaSlicer/issues/3554

@GeekDad63 Brim per object has been requested in https://github.com/prusa3d/PrusaSlicer/issues/286

@nils-kal How do you expect a
>possibility to place custom brims with a modifier.

to work? 

For helper objects acting as brim you can check the Prusa Forum.
e.g.: https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/how-to-print-brim-ears/

Incendium87, 2021-01-26:
Came to post this request. Hopefully it does get completed at some point. This would make things easier for a lot of my models.

ozgurgesli, 2021-05-27:
I need this feature as well. I want to leave a print job with only some items that need the brim. Others definitely not, for cosmetic requirements.

PsychProd, 2021-06-24:
Add me to the list of users requesting brim for only certain models!  Thank you so much...

DaveE128, 2021-07-26:
I'd also find this useful.

schillaci, 2021-08-02:
I need this feature as well.

Cyondil, 2021-09-01:
Adding support for this feature

wuftymerguftyguff, 2021-09-24:
I would  like to be able to enable a brim for selective models on the build plate too.

FerretDave, 2021-09-24:
Would be nice to have an option to selectively add a brim for some items on the build plate, but not others.

shawalli, 2021-09-24:
It'd also be nice to use a modifier to block brim in some sections, like how you can do today with support blockers. An example where this is useful is on the inside of something with close separation, e.g.

[image]

([reference](https://www.thingiverse.com/thing:4911065))

gottesgerl, 2021-11-18:
Would make sense here too.

[image]

I`d love to see this feature

raidolo, 2021-11-25:
+1 also for me it could be a killer feature!

smirgol, 2021-12-18:
+1 from me. 
I need to print narrow pillars that are as wide as the print bed is. They would need a brim to make sure they don't warp, but I cannot add them, as the brim would be outside the print bed at the sides. If I could add a blocker to these sides and still have a brim on the remaining two sides, that really would help a lot.

Roedyroll, 2022-01-04:
this would be amazing to be able to easily make production files with multiple parts

fe60, 2022-01-05:
This item is about _brim blockers_ (already requested in #3554) and _custom brim_, not about _brim per object_ (requested in #286 and implemented in [2.4.0](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0) [1]).

Please clearly separate feature requests and avoid _+1_ comments (use :+1: under the (initial) post instead). That way it will be easier for the development team to follow the discussion.

[1] See release notes of [2.4.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1), look for `Brim and skirt enhancements`.

cryoglobe, 2022-10-07:
Hello 

Could definitely help a lot for "print in place" , or "flexible" models ...  to get the benefit of the brim , but not blocking the "articulated" parts ( see the yellow arrows in the picture )

[image]

joelmoliveira, 2022-12-08:
i would like to see this feature too.

redPanther, 2023-02-20:
would love to see this happening!

Snuff1eupagus, 2024-05-02:
Some of the posts in this thread are older, however this has been possible in PrusaSlicer since Sept 1, 2021. for those that are after that time.

cryoglobe, 2024-05-03:
> Some of the posts in this thread are older, however this has been possible in PrusaSlicer since Sept 1, 2021. for those that posted after that time.
> 
> ![01](https://private-user-images.githubusercontent.com/95652334/327593617-6dec4847-5d48-46c8-8ac6-5f187b26f008.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ3NjcxNDksIm5iZiI6MTcxNDc2Njg0OSwicGF0aCI6Ii85NTY1MjMzNC8zMjc1OTM2MTctNmRlYzQ4NDctNWQ0OC00NmM4LThhYzYtNWYxODdiMjZmMDA4LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTAzVDIwMDcyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM0OGUzYmI0ZWMzM2QxNDBlMTlkYzU0MzYwNDdmNWZkY2ZmZmMwMmY1NmQwYTQ0MzE2NDU3NDU3NzM4N2E1ZjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.jmBRYt49zRrccAV9IzL2ch4gPm3Du-EjQOUiXjvZZYQ)
> 
> ![02](https://private-user-images.githubusercontent.com/95652334/327593692-c350a167-3709-456c-9807-98ebd2c77e4d.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ3NjcxNDksIm5iZiI6MTcxNDc2Njg0OSwicGF0aCI6Ii85NTY1MjMzNC8zMjc1OTM2OTItYzM1MGExNjctMzcwOS00NTZjLTk4MDctOThlYmQyYzc3ZTRkLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTAzVDIwMDcyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE1Y2IyYTVhZmYzNTY1YjA2OTY1ZDI5ZmVkNDM5MGFjMDgwNWRlOGFhZTkyZjA0MDQ2NDVlMTZiMTE4YjlhNGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.VybSJIA9Q1AOcXcTgGTMtDPb00dXIfsrDlWZHq_weDA)

Hello Snuff1eupagus

2021 in Prusa ?   What was the version of Prusa that had this option ?  I remember I had to go to Super Slicer to have this

Snuff1eupagus, 2024-05-04:
@cryoglobe  it ended up just being a  matter of knowing how, this user has show how to do this, and we have been able to since PrusaSlicer 2.4 alpha 1

Hopefully this will help you.

https://www.youtube.com/watch?v=dhsbMmosbLU&t=629s

I learned a lot from this, I hope to see more from this guy.

thegameDB, 2024-05-04:
That’s a workaround, not the feature that is being requested here

Snuff1eupagus, 2024-05-05:
> That’s a workaround, not the feature that is being requested here

Requested feature: Add a modifier to block brims / Add possibility to add custom brims #4744 
Date posted: Sep 12, 2020

This was done, on both fronts, block brims & Add possibility to add custom brims.

 You call it a work around, I call it powerful and easy to use even more powerful then the SuperSlicer implementation.
and easier then using a modifier. 

how? If I might ask? do you think this should be implemented?

smirgol, 2024-05-05:
I admit that I didn't know about this way of adding brims. That would cover the "Add possibility to add custom brims" part of this feature request, which, for the specific issue I had, would have solved the problem.

Still, it would make things easier and more intuitive in my humble opinion if we could do it the other way around: Use standard brim feature and add blockers where they are not needed. How this can possibly be implemented is not up to the feature request to figure out.

If that is feasible at all is up to the devs, but thank you for sharing your opinion. :)

shawalli, 2024-05-06:
> > That’s a workaround, not the feature that is being requested here
> 
> My reply: Nonsense.
> 
> Requested feature: Add a modifier to block brims / Add possibility to add custom brims #4744 Date posted: Sep 12, 2020
> 
> On Sept 1, 2021: This was done, on both fronts, block brims & Add possibility to add custom brims.
> 
> You call it a work around, I call it powerful and easy to use even more powerful then the SuperSlicer implementation. and easier then using a modifier. frankly easier then using the SuperSlicer implementation as well, with far more control.
> 
> how? If I might ask? do you think this should be implemented?

This is absolutely a workaround and not a feature. A feature would be a tool/process that allows a user to quickly and easily put a brim in specific areas.

With your analogy, you don't even need brims at all. Just create a 1-layer shape wherever you want the brim and throw out all that brim logic.

Snuff1eupagus, 2024-05-13:
@shawalli when you grow up and discovery what can actually be done in PrusaSlicer you may "Start" to realize how immature your statements sound. I wish you luck, it seems like you have a few years before getting there.

thegameDB, 2024-05-13:
Condescending asshole much, @Snuff1eupagus?

Snuff1eupagus, 2024-05-13:
> Condescending asshole much, @Snuff1eupagus?

I had hoped for a more intelligent response. But you can only give what you can give. 

Regards

Aneurin, 2024-05-13:
I can't obviously find this on mobile but does anyone know who to ping that can moderate these comments? It would be good to get snuffle's abusive comments deleted, and then maybe a 6 month ban from the project would be appropriate?

Snuff1eupagus, 2024-05-15:
I've looked at each and everyone of you, Aneurin, thegameDB, shawalli, not a single one of you has contributed to the community in any way, yet you all act as though what you have to say matters. this is sad to me and I'm sure to others. Show at least one thing, that you have contributed to us "the community" that is worth your opinion.  How would you like if that was Opposed and you were asked to be band for your attempt to show users what they have had at their fingertips. you are the losers who should be band for your opinions. Yet you accuse and insult others that have offered usable valid information about PrusaSlicer. thinks that you didn't know, are you embarrassed? you should be, clearly you don't understand what can be done.  what can happen beyond that ? you, none of you have offered a thing, who are you? what can you offer? , besides insults and no understanding?

Snuff1eupagus, 2024-08-12:
@smirgol

> I admit that I didn't know about this way of adding brims. That would cover the "Add possibility to add custom brims" part of this feature request, which, for the specific issue I had, would have [...]

This was just released, perhaps this will help fill your need.

https://www.youtube.com/watch?v=6CRqNk_JUj4

I hope it helps.

closed 2026-09-29 by prusaslicer-bot[bot]