prusa3d/PrusaSlicer #4746: First layer width based on percent has bad algorithm

gnydick, 2020-09-12:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 Pro for Workstation 64bit

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
the algorithm uses the first layer height as the multiplier for the percentage. That doesn't effectively manage anything. the most popular first layer height, by far, is **0.2**.  With a **0.4 nozzle**, the default calculation gives you a **0.4** width. That is not a good width. It's always better for your width to be a bit wider than the diameter of your nozzle opening.

This can be worked around with a fixed value that is used for a printing profile for your printer with a **0.4** nozzle, but I have a multi-tool printer with different diameter nozzles.

wavexx, 2020-09-13:
What would you propose as an alternative?

rtyr, 2021-02-24:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2829. Closing.

closed 2021-02-24 by rtyr