prusa3d/PrusaSlicer #4750: Unable to manually add support points

aguskov1987, 2020-09-13:
### Version
2.2.0 release

### Operating system type + version
Windows 10 Home 64 bit

### 3D printer brand / version + firmware version (if known)
Prusa SL1

### Behavior
The problem is relevant in the attached file. Open the file and generate supports with default settings (step 1). Now try manually adding support points: you won't be able to add any on the model surface except for a small circular patch on the bottom of the model (step 2). The model is rather dense -> it is most likely the cause of the bug.

[image]
[image]

Not a feature request

#### Project File (.3MF) where problem occurs
[Rook.zip](https://github.com/prusa3d/PrusaSlicer/files/5214607/Rook.zip)

lukasmatena, 2020-09-14:
Thanks, I can reproduce it. I haven't looked into it yet, but I believe that it is somehow connected to the HUGE scaling applied to the model (50000 %). PrusaSlicer should be able to cope, though.. It is worth investigating.

aguskov1987, 2020-09-15:
@lukasmatena 
Yes, it is the scaling. I up-scaled the model in Blender and re-imported the slt. It worked fine.

bubnikv, 2021-09-22:
Fixed in PrusaSlicer 2.4.0-alpha1

closed 2021-09-22 by bubnikv