prusa3d/PrusaSlicer #4752: Import Prusa Filament Profile to Ideamaker

SJung2020, 2020-09-14:
Dear Prusa Slicer Team 

### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-09-19:
>I have bought a lot of Prusa Filaments from your store and would like to add the different filament profiles to the slicer software of Raise 3D Ideamaker. What is the best way to do this?
> How can I aslo import the different printer profiles of prusa in Ideamaker or how can I import the Raise 3D Pro plus printer into Prusa Slicer?

It is not possible to automatically import printer profiles from PrusaSlicer to Ideamaker or from Ideamaker to PrusaSlicer. I am afraid you will have to create the profiles manually (based on the settings from source slicer) or import some community profiles prepared specifically for your printer.

We are integrating new 3rd party printer profiles to the PrusaSlicer based on the community contributions. We will gladly add Raise 3D Pro printer if anyone in the community provides profiles for it.

closed 2020-09-24 by rtyr

SJung2020, 2020-09-24:
How can I do it: Dont know what I have to do

rtyr, 2020-09-24:
In that case I think the best approach will be to ask someone in RAISE3D community to create PrusaSlicer profile for your printer. Creating printer profile requires certain level of knowledge and also some amount of experimenting, especially with dual extruder printer like this. There is also a possibility, that your printer requires some vendor specific feature, which is not available in different slicers.

I checked Ideamaker profile for RAISE3D Pro2 and created some draft PS profile based on it. It is just very basic profile, which you (or someone else) can use to start creating/tuning proper profile. It may not work properly and possibly can even damage your printer, so **use it at your own risk**.

File->Import->Import configuration bundle...
[RAISE3D_DRAFT_PrusaSlicer_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/5279108/RAISE3D_DRAFT_PrusaSlicer_bundle.zip)

Our tech support can help with some basic PrusaSlicer settings (most likely not with RAISE3D specific settings). As for the Ideamaker, I am afraid you have to ask RAISE3D support for help with their slicer.