prusa3d/PrusaSlicer #4756: "Empty Layer" error is shown too often/early

Technofrikus, 2020-09-15:
### Version
2.2.0+ stable

### Operating system type + version
MacOS 10.15.6

### 3D printer brand / version + firmware version (if known)
Mk3s

### Behavior
When opening/importing models they sometimes are at an angle or the wrong way up, because they were modeled that way. But then I get an error about empty layers, every time. But I havent even begun to position the model. I dont want to print yet.

I think this warning should have an option to disable it permanently or should maybe be shown when exporting the G-Code. Not immediately after importing.

foreachthing, 2020-09-15:
How about a note in a statusbar? I hate that dialogbox (and all others as well)!

lukasmatena, 2020-09-15:
This is already solved in current `master`. The error was turned into a warning so it can be dismissed by the user and does not prevent gcode generation. It is shown as a non-modal box in the corner, similar to what I posted in https://github.com/prusa3d/PrusaSlicer/issues/4606.
The warning will show even when you just import or drag the object - that's the purpose of background processing. But it should not be obtrusive.

Closing.

closed 2020-09-15 by lukasmatena