prusa3d/PrusaSlicer #4758: Q: G-code viewer and ENABLE_GCODE_VIEWER

KimmoHop, 2020-09-15:
Version: 2.3.0-alpha0

Is there any information (documentation, discussion, issue) about ENABLE_GCODE_VIEWER - what it does, what does it change?

Is it possible to build and install G-code previewer?

enricoturri1966, 2020-09-16:
ENABLE_GCODE_VIEWER is technology which embeds all the code regarding the new gcode viewer, both as preview in the regular PrusaSlicer and as standalone application.
As all the other technologies, it is a temporary solution which allow to quickly switch it off, removing it from the code base, if needed.
It will be later removed (set as default) when it will be judged reliable enough.

KimmoHop, 2020-09-17:
It's curious, that there are lines for height
`;HEIGHT:0.2`
but no lines for layer
`;LAYER:0`
like Cura and IdeaMaker do.

Also types like
`;TYPE:External perimeter`
are better but not the same as in two other slicers
`;TYPE:WALL-OUTER`

Due to different output, G-Code Viewer can show only PS output correctly and IdeaMaker (continues to) not show PS output correctly. Unless there is strong reason for different output, I'd suggest using same type names as in Cura and IdeaMaker type names) for compatibility.

lukasmatena, 2024-02-18:
Closing, outdated.

closed 2024-02-18 by lukasmatena