prusa3d/PrusaSlicer #4760: SLA test source code: slicing resolution used before building support tree

Megidd, 2020-09-16:
In SLA test source code, slicing is done here:

https://github.com/prusa3d/PrusaSlicer/blob/b93c95274f99abc23080aea0abc5e7007c6b2332/tests/sla_print/sla_test_utils.cpp#L106

... right before building support tree here:

https://github.com/prusa3d/PrusaSlicer/blob/b93c95274f99abc23080aea0abc5e7007c6b2332/tests/sla_print/sla_test_utils.cpp#L148

I looked at the slicing code, but I couldn't figure out what _2D pixel resolution_ is used by default to create slices/layers. I wonder if I'm missing something. thanks! 😃

tamasmeszaros, 2020-09-21:
Hello! The test is only concerned with generating the support tree mesh, slices in this test are 2D polygons and not a raster, so there is no need for a pixel resolution.

Megidd, 2020-09-21:
@tamasmeszaros Makes sense! 😃  Thanks!

closed 2020-09-21 by Megidd