prusa3d/PrusaSlicer #4766: Option of visualization with edges

jokohoko, 2020-09-18:
### Version
2.2.0

### Operating system type + version
Windows 10

### Behavior
An option to show edges in the view would be great. To see errors or details in the parts i now always open the parts in a 3D-tool first to have some good details.
[image]
3D-tool width edges
[image]
PrusaSlicer without edges

New feature request!

zakutin, 2021-06-04:
This feature would be really helpful! For example, I'm using the top view, during the arrangement of the parts on the building platform. Often I have some markers and elements embossed on the part, which are _not visible_ from the top.
[image]
[image]

zakutin, 2026-09-18:
still required