prusa3d/PrusaSlicer #4767: (Feature Request) Minimum Extrusion Length

Skelten123, 2020-09-19:
Feature Request:
Minimum extrusion length modifier.

This could reduce print time and decrease the size of gcode files. I would imagine that the modifier would affect the minimum distance the extruder travels between the start and end of a single line of extruded filament.

This would probably go well with variable extrusion width to minimize the total extrusion amounts.

Small extrusion lengths can be especially hard with the first layers where the filament is not connected with others. This can cause layer adhesion issues where the nozzle bumps the small pieces and then a runaway effect happens with a build-up of small pieces.

carlosjln, 2020-11-13:
I was just looking for this feature... sighs, The most importan part of this is to prevent the slicer from generating extrusions that are just dots. That's what's happening on my current print, the extruder completes some perimeters then travels to another corner of the object just to deposit a single dot. These moves are just a waste of time and could cause more harm than anything else.

YairHH, 2021-01-22:
This is an absolute requirement for materials like ABS or PETG, very small extrusions specially if they happen consecutively can diminish the pressure inside the nozzle to the point where restarting a long path will have under extrusion at the beginning.

Skelten123, 2021-01-23:
I did some looking around on GitHub and found this to be similar but still different from the feature requests found in:
#1806 #2691 #2780
This appears to be a setting that will help considerably in lots of areas during printing and could help with reliability.

boromyr, 2021-02-20:
I also subscribe to this request, punctual extrusions waste a lot of time and dirty the nozzle and the print.

JonBerlin, 2021-11-29:
Please Prusa add this feature! This is a must for printing organic textures.

Lukrative525, 2022-11-29:
Just adding another voice of support for the addition of this feature.

timur-tabi, 2023-06-16:
I would also love to see this feature.  I'm printing a box with interior walls, and with the Arachne engine, PrusaSlicer insists on filling in the corners with a small blob.  Unfortunately, PS travels each layer very inefficiently, and these blobs double the print time and provide no real benefit.

[image]

Christopher-Dauchter, 2023-07-16:
Would really love to see this feature. I've run into this issue in nearly 80% of my prints.

Googliola, 2023-07-29:
One more pretty please!

Googliola, 2023-07-29:
Arachne perimeter generator offers some powerful (officially undocumented) features that might help to workaround the zist issue (which is a nightmare when printing with soft TPU). Go to Print Settings > Advanced > APG to find the settings referenced below.

Default settings:
[image]

1 and 2 should stick to their neighbours
[image]

2 is resolved, 1 remains a minor issue
[image]

While Arachne perimeter generator might solve the minimal lenght issue, it is a workaround after all. I still wish there was an easier way to cope with baby blobs.

IonNight, 2023-08-21:
Its weird that "Minimum Extrusion Length" is not yet implementet. Im having a print where it makes so many small extrusions the print time went from 55 minutes with 2 perimeters to 61 minutes with 3 perimeters to 94 minutes with 4 perimeters.

Christopher-Dauchter, 2024-08-15:
Been about a year since I last requested this. I am still very interested in this feature. The Arachne minimum feature does not fix all small path problems.

hugo-id, 2024-10-25:
Here is a file I'm working on;
[image]
With two perimeters (required elsewhere), print time is **9 minutes**

If I reduce to 1 perimeter:
[image]
**Print time is only 5 minutes**

This isn't really acceptable as now other parts of the print will be weakened, and the slicer is still generating spurious solid infill blobs (visible elsewhere)

But it's basically **halving** the print time, for a tiny print!

The amount of time this absolutely basic feature could save is _insane_ and I don't understand why Prusa is dragging its feet on this one.

Just add a filter that if length < X, the extrusion is removed.

Note:
I can't even work around this issue as due #3370 , I can't use modifiers to selectively enforce the perimeters where they're needed

Spaun39, 2024-11-01:
It's been over four years and this feature is still missing, crazy.

timur-tabi, 2024-11-01:
Crazy, but not unexpected from Prusa.  It's easy to find hundreds of neglected bug reports in this issue tracker.

dgudim, 2024-11-16:
I spent like 20min trying to find this option, but it's not even implemented ... in 4 years!!!!

jwidess, 2025-03-02:
Wow, I can't believe this still hasn't been added! [SuperSlicer has had this since 2022](https://github.com/supermerill/SuperSlicer/issues/278), and it's still a problem I run into regularly causing jams on complex first layers with lots of microscopic dot extrusions.

Spaun39, 2026-09-02:
The problem is still relevant and I'm still waiting for a solution to appear.

IonNight, 2026-09-02:
No you Prusa bot. I dunno why anyone not would want this function. It helps all

Googliola, 2026-09-03:
On some prints, this is the major cause for failure due to the endless retracts within short time-frame. Please implement.

Jan-Soustruznik, 2026-09-04:
Hello, I have moved this feature request to GitHub Discussions in accordance with our [policy](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy).