prusa3d/PrusaSlicer #4768: Infill before perimeters on random layers while set off in settings

KimmoHop, 2020-09-19:
### Version
version_2.3.0-alpha0-894-ga09cb3c79

https://github.com/KimmoHop/PrusaSlicer/tree/m117
which is master at 7a10e23470384ba13062a059b48425b86a58b7ed + changes for M117 printing

### Operating system type + version
W10 Home 20H2 19042.508

### 3D printer brand / version + firmware version (if known)
Anycubic i3 Mega + Marlin 1.1.9

### Behavior
After slicing, 1st, 3rd and 4th layers print infill before perimeter while that setting is off:
`; infill_first = 0`
Other layers seem to keep the correct order.
I tried to print the bottom with thinner layers with height range modifier and thought that might have done something, but the same after removing modifier.

In the preview from midway of first layer, infill is done and perimeters has only started.

[image]

#### Project File (.3MF) where problem occurs
[rapu-duct-3.zip](https://github.com/prusa3d/PrusaSlicer/files/5249319/rapu-duct-3.zip)

G-Code clip 1: support -> infill (layer 0)
```
G1 X151.000 Y91.572 E1.72830 ; support material interface
G1 X151.000 Y88.054 E1.83275 ; support material interface
G1 X151.651 Y88.011 F6000.000 ; move to first support material point
;TYPE:Support material
G1 F1200.000
G1 X151.651 Y85.552 E1.93642 ; support material
G1 E0.43642 F1800.00000 ; retract
G92 E0 ; reset extrusion distance
G1 Z0.280 F6000.000 ; lift Z
G1 X155.766 Y79.876 ; move to first infill point
G1 Z0.200 ; restore layer Z
G1 E1.50000 F1800.00000 ; unretract
;TYPE:Solid infill
G1 F1200.000
G1 X154.555 Y78.664 E1.57260 ; infill
G1 X154.107 Y78.258 E1.59821 ; infill
G1 X153.607 Y77.887 E1.62459 ; infill
```

G-Code clip 2: end of infill -> perimeter
```
G1 X129.044 Y130.259 E18.81714 ; infill
G1 X126.670 Y128.604 E18.93954 ; infill
G1 X124.155 Y126.089 E19.09000 ; infill
G1 E17.59000 F1800.00000 ; retract
G92 E0 ; reset extrusion distance
G1 Z0.280 F6000.000 ; lift Z
G1 X107.627 Y105.067 ; move to first perimeter point
G1 Z0.200 ; restore layer Z
G1 E1.50000 F1800.00000 ; unretract
;TYPE:External perimeter
G1 F1200.000
G1 X112.236 Y102.331 E1.59257 ; perimeter
G1 X112.271 Y102.287 E1.59374 ; perimeter
G1 X112.306 Y102.243 E1.59509 ; perimeter
```

rtyr, 2020-09-19:
Reproduced. It is OK with "detect thin walls" disabled.

pseudex, 2020-12-12:
[image]

I can't get perimeters first at all to work. even on a simple shape.

Version: PrusaSlicer-2.3.0-beta2+linux-x64-202012051122.AppImage

ralf456, 2021-04-29:
I think I have found the issue!

During the `traverse_loops` the thin walls are added and then the path is determined using the _greedy algorithm_ over all elements. Here the start_near `zero_point` value is `(0, 0)` which causes the thin walls to be printed before the perimeter, if they are closer to zero as the start point of the perimeter.

https://github.com/prusa3d/PrusaSlicer/blob/77f016246fe1e7aa88f16958e944da81a161c362/src/libslic3r/PerimeterGenerator.cpp#L200-L201	
	
Because the first point of the perimeter is a thin wall, `point_inside_surface` fails in the G-Code generator, and the perimeter is packed into the last `islands`. So in my example, the perimeter is in `islands[1]` and the infill is in `islands[0]`.

https://github.com/prusa3d/PrusaSlicer/blob/77f016246fe1e7aa88f16958e944da81a161c362/src/libslic3r/GCode.cpp#L2015-L2023

As a result, when putting them together, the infill in the first loop is printed and the perimeters in the second.

  https://github.com/prusa3d/PrusaSlicer/blob/77f016246fe1e7aa88f16958e944da81a161c362/src/libslic3r/GCode.cpp#L2116-L2128     

This results in printing the infill before the perimeter.

Attached is another small file where the problem occurs in layer 18 and 21.
[Test_Gcode_stl_config.zip](https://github.com/prusa3d/PrusaSlicer/files/6399670/Test_Gcode_stl_config.zip)

[image]

KimmoHop, 2021-04-29:
I tested with 2.4.0-alpha (d7565d96bf99caf969439961b9d87257716c4cb0) and disabling thin walls creates gcode in correct order.

Thin walls is funny, sometimes it makes print better, sometimes worse. Maybe behaviour like in SuperSlicer could be used?

closed 2026-09-29 by prusaslicer-bot[bot]