prusa3d/PrusaSlicer #4769: On Linux, "Remove Device" corrupts the SD 90% if the times (missing "sync")

zevix, 2020-09-19:
### Version
2.2.0+linux-x64

### Operating system type + version
Fedora 32

### 3D printer brand / version + firmware version (if known)
i3MK3

### Behavior
Got many SD filesystem corruptions when using "Export to SD" and "Remove Device" from the UI
Since they responded suspiciously fast - both Export and Remove returned immediately, I figured out the "Remove" does not wait for the IO to complete. 

I verified this by:
* Exporting to SD
* Switching to a terminal and running "sync"
* Waiting for sync to complete (took a few seconds)
* Switching back to Slicer, clicking "Eject"

Using the procedure above above eliminates all SD corruptions

Looking at the code. looks like RemovableDeviceManager needs to call  sync() or syncfs() before ejecting

DanielJoyce, 2020-10-01:
It should just call sync automatically after finishing writing the file if the media is removable. It's not going to be hitting that flash drive over and over, so the performance hit is not an issue, it should write the file, then call sync immediately.

bubnikv, 2020-10-01:
> Looking at the code. looks like RemovableDeviceManager needs to call sync() or syncfs() before ejecting

I don't think so. The unmount kernel call shall take care of that.

I suspect your SD card or flash drive is wonky. Some of the SD cards or flash drives are notoriously bad, they survive just a handful of writes. Some drivers are so bad that they do basically no wear leveling. Nobody knows when the driver will actually flush its caches if it has any.

zevix, 2020-10-09:
> I don't think so. The unmount kernel call shall take care of that.
Does not align with me experience, see also https://unix.stackexchange.com/questions/345917/does-umount-calls-sync-to-complete-any-pending-writes 

> I suspect your SD card or flash drive is wonky

I would not blame the SD card. I checked with more than one card, and my main card is a high-end SanDisk Extreme-Pro and works flawlessly except this specific scenario. 

Furthermore, even if it was an SD controller bug (and it's not), the end user should not care...

If I had the right dev env I would have submitted a fix myself... Would really appreciate a fix from someone more capable of verifying ....

bubnikv, 2021-01-07:
Frankly if there was a problem in PrusaSlicer, I expect mutliple reports of the issue.

closed 2026-09-29 by prusaslicer-bot[bot]