prusa3d/PrusaSlicer #4771: PrusaSlicer crashing, does not like to open unless it is a stock profile.

Spankyty, 2020-09-19:
### Version
2.2.0

### Windows 10 + 1903
OS build 18362.1082
PrusaSlicer installed on D: Drive and my AppData folder is on the C: Drive. Dunno if that makes a difference.

### Behavior
* PrusaSlicer does not like to open unless it is a profile that is stock (MK2.5S, Ender 3 Pro, MK2.5 MMU2S. Work fine.)
* If you keep opening PrusaSlicer at some point it will stay open and not crash. (Most times it crashes)
* Create a custom printer named CR-10 V2
  * Marlin Firmware
  * set the build size 300x300 and origin to 0,0. 
  * load the texture and model.
  * Nozzle diameter 0.4mm
  * Filament Diameter 1.75mm
  * Extrusion Temp: 215C
  * Bed Temp: 60C
  * Click Finish
* _Expected Results_
  * A working CR10 Profile.
* _Actual Results_
  * Random crashes when selected custom profile.
  * [Video](https://github.com/prusa3d/PrusaSlicer/files/5250131/2020-09-19.11-24-39.zip)
  * [PrusaSlicer Config Folder.zip](https://github.com/prusa3d/PrusaSlicer/files/5250133/PrusaSlicerRoamingFolder.zip)
  * Cannot upload the Program Folder. Too big.

rtyr, 2020-09-19:
It seems to be caused by your texture png. Can you try it without the texture?
FYI @enricoturri1966

Spankyty, 2020-09-19:
> It seems to be caused by your texture png. Can you try it without the texture?
> FYI @enricoturri1966

Yup, that was the issue. Thank you! Do you know why it was caused by the png?

closed 2020-09-20 by Spankyty