prusa3d/PrusaSlicer #4772: No Flash Drive available on PopOS / Ubuntu in PrusaSlicer

keen101, 2020-09-19:
### Version
_Version of PrusaSlicer 2.2.0-1_

### Operating system type + version
_What OS are you using, and state any version #s_
PopOS 20.04 LTS (based on Ubuntu) 64-bit

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa Mini Stock - firmware not applicable to this problem

### Behavior
* _Describe the problem_
The Prusa Slicer on PopOS 20.04 (Ubuntu) installed from flatpak seems to lack access to the USB Flash Drives for normal g-code export. Manually browsing in the prusa slicer menu seems to lack access to /media on the Linux filesystem. I think some sort of root permissions are not set properly.

* _Steps needed to reproduce the problem_
Install Pusa Slicer on PopOS 20.04 via the "Pop!_Shop"
* _Expected Results_
Ability to save gcode to usb-flash-drive media
* _Actual Results_
no usb drive available

lukasmatena, 2020-09-20:
Can you please try with our AppImage (https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.2.0/PrusaSlicer-2.2.0+linux-x64-202003211856.AppImage) on your system?

keen101, 2020-09-20:
Yes, the PrusaSlicer-2.2.0+linux-x64-202003211856.AppImage works with my flash drive.

The only downside is i would prefer a .deb or similar installable package so i could have an integrated .desktop launcher icon and also have my system automatically associate .stl files with prusaslicer. But i am assuming this new "appimage" is some sort of image like flatpak or snap images that are compatible on many different linux systems. Perhaps you can fix the flatpak image that is in the PopOS repositories?

lukasmatena, 2020-09-20:
@keen101 
Linux world is open, which is why there are loads of different distros. This creates a need to be able to deploy applications in distro-agnostic manned. And again, because Linux world is open, this issue is being solved in multiple independent ways. Snap, Flatpak and AppImage are three different answers to that. And AppImage is not "new", I believe it is in fact the oldest of the three.

Prusa Reasearch has no resources to maintain major distribution packages as well as Snap, Flatpak and AppImage packages, not to mention that majority of PS users do not even run Linux. That's why we provide an AppImage (which is dead simple and just works) and leave the others to the community that needs them so much.

Flatpak package was AFAIK created by @xarbit, who is the right person to have a look at this issue. By the way, it is quirks like this why we are not at all keen on taking over the maintenance of these packages. Keeping PrusaSlicer at least buildable on all the three platforms is challenging enough.

Also see e.g. #1124 or #3064.

keen101, 2020-09-29:
> @keen101
> Linux world is open, which is why there are loads of different distros. This creates a need to be able to deploy applications in distro-agnostic manned. And again, because Linux world is [...]

Thanks @lukasmatena ,

A fair response and explanation. I figured there was a fair chance you were probably not the ones maintaining the flatpack image anyway, but i was not sure. @xarbit perhaps you can look into it.

After using the official AppImage package it seems to work very well and i have no issues with it. I can see why Prusa would want to maintain the PrusaSlicer in the manner. I was not aware of cross-distro AppImage packages until now, only Snap and Flatpack images. 

The only minor inconvenience is that there is no built in .desktop shortcut file that I can call up when i press my "windows" key and type "prusa" into my search bar for quick easy launching. Do you know of any .desktop launchers that others have already created that I could then copy to my system manually? I recently had to do something similar for my Autodesk EagleCAD program, so it's something that could be done. I would just need the PrusaSlicer icon image after that to make the launcher look nice.

Otherwise thanks for the reply. I guess you can mark this issue as closed.
Otherwise thanks for the he

lukasmatena, 2020-09-29:
> The only minor inconvenience is that there is no built in .desktop shortcut file that I can call up when i press my "windows" key and type "prusa" into my search bar for quick easy launching.

If that's the only problem, I would just write a shell script to run it and placed it in PATH (e.g. `~/.local/bin/prusa-slicer.sh`). That might be enough.

kirameister, 2021-06-04:
Hello, I got the same issue and it seems to be an expected behavior of Flatpak package. Following is what I did: 

1. Install `Flatseal` from Pop!_Shop
2. Launch `Flatseal`, select `PrusaSlicer` and turn on "All system files" under "Filesystem"
3. Launch PrusaSlicer

After slicing your STL file, you should be able to see "Export to SD card / Flash drive" icon next to the "Export G-code" button.

migxmac, 2021-07-20:
@kirameister That did it. Thank you.

bubnikv, 2021-07-29:
> If that's the only problem, I would just write a shell script to run it and placed it in PATH (e.g. `~/.local/bin/prusa-slicer.sh`). That might be enough.

Actually we will bring desktop integration with PrusaSlicer 2.4.0-alpha1. The desktop integration will be optionally invoked at the end of the installation wizard and from the menu.

xarbit, 2021-09-02:
there is a update building for PrusaSlicer 2.3.3 with more permissions. I hope this fixes the issues. 

https://flathub.org/builds/#/builders/32/builds/58287

xarbit, 2021-12-12:
@bubnikv @lukasmatena this can be closed, or?

migxmac, 2021-12-12:
Yes, please close.
Thank you

closed 2021-12-12 by bubnikv

thekarinka, 2022-11-14:
See this https://github.com/ivocavalcante/PrusaSlicer-snap, it's need to connect removable media for snap.