prusa3d/PrusaSlicer #4774: Multiple bridging layers on top of each other

chgi, 2020-09-20:
### Version
2.2.0+win64
2.3.0-alpha0-902-gf2951b53c (f2951b53c0f958dd443d44b2f6f0b687593153ed)

### Operating system type + version
Windows 10 Pro 1909 64bit

### 3D printer brand / version + firmware version (if known)
Seckit SK-Go with Marlin 2.0.6.1

### Behavior

I believe I have found an edge case where Prusaslicer wrongly uses bridge flow in an area that is already bridged, leading to heavy overextrusion.
By my calculations I get as much of 12 layers worth of filament where only 2 layers should be (0.6mm nozzle, 0.1mm layers).
This in turn leads to the nozzle scraping over the area, making a horrible rattling noise and bending the print bed.
It happens on the underside of nearly horizontal sloped walls, that are supported on two sides.

I see this behaviour in the preview for 0.1mm layer height in
- PrusaSlicer 2.2.0 with slopes from about 2 degree to 5 degrees and
- PrusaSlicer 2.3.0-alpha0-902-gf2951b53c from about 1 degree to 7 degrees.

Different layer heights somewhat shift the degree of inclination where it appears.

* _Steps needed to reproduce the problem_

Slice the included 3mf file.
Inspect the bridged overhangs. It is pretty difficult to see.
One must look at single layers from above and below.
Hopefully the screenshots show what I mean.

* _Expected Results_

There should be no overlapping bridge layers (I think).

* _Actual Results_

Up to 2 Bridge layers are printed on top of an area that is already (mostly) bridged by bridged perimeters.

Full model view, all visible bridge areas exhibit the problem, below zoomed are circled
[image]

Zoom in, 3 layers, bottom view
[image]

Zoom in, 3 layers, top view
[image]

Zoom in, single layer
[image]

Actual (aborted) print result 1
[image]

Actual (aborted) print result 2
[image]

Actual (aborted) print result 3
[image]

_Is this a new feature request?_ NO

#### Project File (.3MF) where problem occurs

- Soapy3v10.3mf contains the model (a soap holder) where I originally noticed the problem.
- TestObject.3mf is just a standalone test case to illustrate the bahaviour. It contains a slope from 1 to 8 degrees in 1 degree increments.

[demo.zip](https://github.com/prusa3d/PrusaSlicer/files/5251660/demo.zip)

wavexx, 2020-09-21:
See #3320 where there is a model (slipe_test) which highlights this problem

mws-rmain, 2021-04-15:
@chgi - what are you using to get the photos of your parts? Very impressive close-ups!

chgi, 2021-05-09:
@mws-rmain sorry to take so long to answer.
I made those pictures with my phone, a Google (Huawei) Nexus 6P at the time.
No magic either. Good light near the window, go so close that it will just barely be still able to autofocus, then crop to the relevant area.

closed 2026-09-29 by prusaslicer-bot[bot]