prusa3d/PrusaSlicer #4777: Per-printer post-processing script

wavexx, 2020-09-21:
### Version
6cdb19971fa755032d40c05546c14b348caa47f9

### Behavior
It would be useful to have a per-printer gcode post-processing script, in the same way we do currently for print profiles.

As I started to use a post-processing script to adjust some machine limits during the print, I find that I want to use the same script for all print profiles for that machine.

If both a print and a printer list of post-processing scripts exist, both should run. Essentially, we could combine the lists and run them in succession.

Which one should be run first (print or printer) is something I'm still not fully sure about, since both orders have advantages/disadvantages depending on the outcome. I'd love if somebody that is currently using post-processing scripts would join the argument here :)

_Is this a new feature request?_

Yes

foreachthing, 2020-09-22:
Or, you could use this line (or something to fit your needs) in `Start GCode`:
`; Printer_Settings_ID: [printer_settings_id]`

This will then output something like this in your GCode: `; Printer_Settings_ID: MyUM2-Custom-Plate`.
Then you only need to check for this string in your post-processing script and do you printer specific thing. This way you only have one post-processing-script to maintain as well - maybe you like that, maybe you don't  ;-)

To get the names of each profile (Print, Filament and Printer), use any of those placeholders:
`print_settings_id`, `filament_settings_id` and `printer_settings_id`.

Here are more possible placeholders. https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-placeholders-(a-copy-of-the-mauk.cc-page)

wavexx, 2020-09-26:
> Or, you could use this line (or something to fit your needs) in `Start GCode`:
> `; Printer_Settings_ID: [printer_settings_id]`

That sounds like a way to avoid having to run more than a single gcode script, which is nice. But I guess you'd still need a per-printer gcode script in order to actually "use" this.

closed 2026-09-29 by prusaslicer-bot[bot]