prusa3d/PrusaSlicer #4780: Infill solid instead of sparse for smaller objects (puzzling settings/ bug?)

filipgoc, 2020-09-21:
### Version
Version 2.2.0+

### Operating system type + version
10.13.6

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
- Set Print Settings, Filament Settings to _default_ and Printer settings to _default FFF_
- Try to slice attached 10x10x10mm box stl
- The **infill is completely solid** even though there are only 3 perimeters and 10% infill 😕 

I'm not sure if this is a bug or some weird setting somewhere. Any pointers appreciated!
Thanks!

**Screenshot**
[image]

#### Project File (.3MF) where problem occurs
[weird-infill.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5258275/weird-infill.3mf.zip)

neophyl, 2020-09-22:
Under Print Settings>Infill>Advanced>Solid Infill threshold area you have it set to 70mm2.  That means any area below this value will be filled with solid infill.  Reduce it to (something like 10) and the smaller cube will also use normal infill.

rtyr, 2020-10-02:
Explained. Closing.

closed 2020-10-02 by rtyr

filipgoc, 2020-10-02:
> Under Print Settings>Infill>Advanced>Solid Infill threshold area you have it set to 70mm2. That means any area below this value will be filled with solid infill. Reduce it to (something like 10) and the smaller cube will also use normal infill.

YES This was it. Thank you so much!
(I forgot to click the 'comment' button and it got stuck in drafts.)