prusa3d/PrusaSlicer #4781: Extruder select is broken after selecting the already selected extruder

plampix, 2020-09-21:
### Version
version_2.3.0-alpha0-263-g171acf094

### Operating system type + version
Debian GNU/Linux 10

### 3D printer brand / version + firmware version (if known)
MK3S Stock

### Behavior
When I try to select an extruder in the parts form, and select the same extruder that was already selected (standard->standar, 1->1 2->2 etc) the selector closes, and I won't be able to select another extruder until I restart PrusaSlicer.
To reproduce, select an MMU2(S) printer, create a cube, click on the extruder selector, select 'standard' (which was already selected), and try to change it again (won't work)

This is a bug, not a feature request. It used to work reliably in PrusaSlicer 2.2 (AppImage)

#### Project File (.3MF) where problem occurs
[simple_block_mmu2.zip](https://github.com/prusa3d/PrusaSlicer/files/5258653/simple_block_mmu2.zip)

lukasmatena, 2020-09-22:
I cannot reproduce it on Ubuntu 16.04 (wxWidgets 3.1, GTK2). What wxWidgets and GTK have you built it against? I smell something along those lines.

plampix, 2020-09-22:
I can confirm this was the issue, I tried building it using the system-supplied wxWidgets (3.0.4), and this seems to be the issue. Building with the wxWidgetts in the 'deps' directory fixes the issue.

lukasmatena, 2020-09-24:
We might still try to do something about it, there are people (most importantly package maintainers) who will link against wxWidgets 3.0...

plampix, 2020-09-24:
Ok, that would be nice as it is an important feature for MMU2 owners.

glutorange, 2020-11-21:
I can confirm this issue, too. Please let me know if you need more information to reproduce this issue. 
My PrusaSlicer version: 2.3.0-alpha3.

glutorange, 2020-11-22:
> I can confirm this issue, too. Please let me know if you need more information to reproduce this issue.
> My PrusaSlicer version: 2.3.0-alpha3.

Update: Seems not to happen in alpha4!

plampix, 2020-11-23:
The problem still exists, @glutorange did you build it using wxWidgets 3.0.X?

closed 2022-01-19 by plampix