prusa3d/PrusaSlicer #4782: "cmake -DSLIC3R_WX_STABLE=1" fails on Ubuntu 20.04 "(Could NOT find wxWidgets)"

AJRepo, 2020-09-22:
### Version
 version_2.3.0-alpha0-904-g6cdb19971 
Latest from git ( 6cdb19971fa755032d40c05546c14b348caa47f9 ) 

### Operating system type + version
OS = Ubuntu 20.04.1 LTS 
uname = Linux ao3 5.4.0-47-generic #51-Ubuntu SMP Fri Sep 4 19:50:52 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

I think I have all the required dependencies: 
```
$ dpkg --list | egrep "(zlib)|(libcurl)|(tbb)|(boost)|(eigen)|(glew)|(expat)|(openssl)|(nlopt)|(openvdb)|(libwx)" | grep dev | grep ii | sort | awk '{print $2, $3}'| column -c 2 -t
libboost1.71-dev:amd64                  1.71.0-6ubuntu6
libboost1.71-tools-dev                  1.71.0-6ubuntu6
libboost-all-dev                        1.71.0.0ubuntu2
libboost-atomic1.71-dev:amd64           1.71.0-6ubuntu6
libboost-atomic-dev:amd64               1.71.0.0ubuntu2
libboost-chrono1.71-dev:amd64           1.71.0-6ubuntu6
libboost-chrono-dev:amd64               1.71.0.0ubuntu2
libboost-container1.71-dev:amd64        1.71.0-6ubuntu6
libboost-container-dev:amd64            1.71.0.0ubuntu2
libboost-context1.71-dev:amd64          1.71.0-6ubuntu6
libboost-context-dev:amd64              1.71.0.0ubuntu2
libboost-coroutine1.71-dev:amd64        1.71.0-6ubuntu6
libboost-coroutine-dev:amd64            1.71.0.0ubuntu2
libboost-date-time1.71-dev:amd64        1.71.0-6ubuntu6
libboost-date-time-dev:amd64            1.71.0.0ubuntu2
libboost-dev:amd64                      1.71.0.0ubuntu2
libboost-exception1.71-dev:amd64        1.71.0-6ubuntu6
libboost-exception-dev:amd64            1.71.0.0ubuntu2
libboost-fiber1.71-dev:amd64            1.71.0-6ubuntu6
libboost-fiber-dev:amd64                1.71.0.0ubuntu2
libboost-filesystem1.71-dev:amd64       1.71.0-6ubuntu6
libboost-filesystem-dev:amd64           1.71.0.0ubuntu2
libboost-graph1.71-dev:amd64            1.71.0-6ubuntu6
libboost-graph-dev:amd64                1.71.0.0ubuntu2
libboost-graph-parallel1.71-dev         1.71.0-6ubuntu6
libboost-graph-parallel-dev             1.71.0.0ubuntu2
libboost-iostreams1.71-dev:amd64        1.71.0-6ubuntu6
libboost-iostreams-dev:amd64            1.71.0.0ubuntu2
libboost-locale1.71-dev:amd64           1.71.0-6ubuntu6
libboost-locale-dev:amd64               1.71.0.0ubuntu2
libboost-log1.71-dev                    1.71.0-6ubuntu6
libboost-log-dev                        1.71.0.0ubuntu2
libboost-math1.71-dev:amd64             1.71.0-6ubuntu6
libboost-math-dev:amd64                 1.71.0.0ubuntu2
libboost-mpi1.71-dev                    1.71.0-6ubuntu6
libboost-mpi-dev                        1.71.0.0ubuntu2
libboost-mpi-python1.71-dev             1.71.0-6ubuntu6
libboost-mpi-python-dev                 1.71.0.0ubuntu2
libboost-numpy1.71-dev                  1.71.0-6ubuntu6
libboost-numpy-dev                      1.71.0.0ubuntu2
libboost-program-options1.71-dev:amd64  1.71.0-6ubuntu6
libboost-program-options-dev:amd64      1.71.0.0ubuntu2
libboost-python1.71-dev                 1.71.0-6ubuntu6
libboost-python-dev                     1.71.0.0ubuntu2
libboost-random1.71-dev:amd64           1.71.0-6ubuntu6
libboost-random-dev:amd64               1.71.0.0ubuntu2
libboost-regex1.71-dev:amd64            1.71.0-6ubuntu6
libboost-regex-dev:amd64                1.71.0.0ubuntu2
libboost-serialization1.71-dev:amd64    1.71.0-6ubuntu6
libboost-serialization-dev:amd64        1.71.0.0ubuntu2
libboost-stacktrace1.71-dev:amd64       1.71.0-6ubuntu6
libboost-stacktrace-dev:amd64           1.71.0.0ubuntu2
libboost-system1.71-dev:amd64           1.71.0-6ubuntu6
libboost-system-dev:amd64               1.71.0.0ubuntu2
libboost-test1.71-dev:amd64             1.71.0-6ubuntu6
libboost-test-dev:amd64                 1.71.0.0ubuntu2
libboost-thread1.71-dev:amd64           1.71.0-6ubuntu6
libboost-thread-dev:amd64               1.71.0.0ubuntu2
libboost-timer1.71-dev:amd64            1.71.0-6ubuntu6
[... 143 lines cut ...]

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}
```

_Is this a new feature request?_

No

JacekJagosz, 2020-09-27:
Remember you also lack cereal, try to install that as well.
`-- Could NOT find cereal (missing: cereal_DIR)`

JacekJagosz, 2020-09-27:
I had the same problem and I believe what you need to do is disable precompiled headers, which breaks the build on newer cmake.
`-DSLIC3R_PCH=0`

AJRepo, 2020-09-28:
In regards to cereal not being found... Since it didn't stop cmake, it probably found it before as in issue #3942.  I'm not sure if I installed the library or if in "git update" it added new messages. 

In re-checking the latest cmake build now returns ... 

```
-- Could NOT find cereal (missing: cereal_DIR)
-- Looking for C++ include cereal/cereal.hpp
-- Looking for C++ include cereal/cereal.hpp - found
```

So it looks like it's working. 

For the full cmake: I tried with -DSLIC3R_PCH=0 as 

`cmake -DSLIC3R_WX_STABLE=1 -DCMAKE_BUILD_TYPE=Release  -DSLIC3R_PCH=0 ..`

but it still fails with the error 
"`Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS) (Required is at least version "3.0")`"

I can get cmake to complete successfully by specifying DCMAKE_PREFIX_PATH=... and DSLIC3R_STATIC=1 but I don't see a difference between using  -DSLIC3R_PCH=0 or not. 

I figured out the reason for the error "the error "The program 'prusa-slicer' received an X Window System error." with the final compiled binary. That was because of the installation of a  package that had required a reboot, and I hadn't rebooted. 

So, since I can compile using static WxWidgets-3.1 and WxWidgets-3.0 seems to be deprecated (#4282) I'll close this ticket.

closed 2020-09-28 by AJRepo